borglab/gtsam #1101: ISAM2 Marginalization Issues

gradyrw, 2022-02-15:
## Description

I have been trying to use ISAM2's marginalizeLeaves function and encountered a few issues, particularly on larger problems. I've tracked the issues down as best I can and I believe there are 3 different bugs in the marginalizeLeaves function which are causing problems:
1. There is a missing `nullptr` check which is required in cases where a leaf is also a root (e.g. marginalizing the only node in a graph).
   
2. The vertical block matrix of the clique's conditional is not getting updated properly in cases where the entire clique is not marginalized
   
3. The logic for splitting a clique assumes that keys are ordered in a way such that all the marginalized keys appear before any keys that are being kept. This is not always true.
   
I think there are simple solutions for the first two issues, but I'm not sure about the final one.

## Steps to reproduce

I'm using this code to force the keys that I want to marginalize to be leaves and to then update ISAM2 and marginalize:
```
boost::optional<FastMap<Key, int> > createOrderingConstraints(const ISAM2& isam, const KeyVector& new_keys,
                                                              const KeySet& marginalizable_keys)
{
  if (marginalizable_keys.empty()) {
    return boost::none;
  } else {
    FastMap<Key, int> constrained_keys = FastMap<Key, int>();
    // Generate ordering constraints so that the marginalizable variables will be eliminated first
    // Set all existing and new variables to Group1
    for (const auto& [key, val] : isam.getDelta()) {
      constrained_keys.emplace(key, 1);
    }
    for (const auto& key : new_keys) {
      constrained_keys.emplace(key, 1);
    }
    // And then re-assign the marginalizable variables to Group0 so that they'll all be leaf nodes
    for (const auto& key : marginalizable_keys) {
      constrained_keys.at(key) = 0;
    }
    return constrained_keys;
  }
}

void markAffectedKeys(const Key& key, const ISAM2Clique::shared_ptr& root_clique, KeyList& additional_keys)
{
  std::stack<ISAM2Clique::shared_ptr> frontier;
  frontier.push(root_clique);
  // Basic DFS to find additional keys
  while (!frontier.empty()) {
    // Get the top of the stack
    const ISAM2Clique::shared_ptr clique = frontier.top();
    frontier.pop();
    // Check if we have more keys and children to add
    if (std::find(clique->conditional()->beginParents(), clique->conditional()->endParents(), key) !=
        clique->conditional()->endParents()) {
      for (Key i : clique->conditional()->frontals()) {
        additional_keys.push_back(i);
      }
      for (const ISAM2Clique::shared_ptr& child : clique->children) {
        frontier.push(child);
      }
    }
  }
}

ISAM2Result updateAndMarginalize(const NonlinearFactorGraph& new_factors, const Values& new_values,
                                 const KeySet& marginalizable_keys, ISAM2& isam)
{
  // Force ISAM2 to put marginalizable variables at the beginning
  const boost::optional<FastMap<Key, int> > ordering_constraints =
      createOrderingConstraints(isam, new_values.keys(), marginalizable_keys);

  // Mark additional keys between the marginalized keys and the leaves
  KeyList additional_marked_keys;
  for (Key key : marginalizable_keys) {
[... 417 lines cut ...]
  // Optimize the graph
  auto result = updateAndMarginalize(factors, values, {}, isam);
  auto estimate = isam.calculateBestEstimate();

  // Get the list of keys
  std::vector<Key> key_list(GRID_DIM * GRID_DIM);
  std::iota(key_list.begin(), key_list.end(), 0);

  // Shuffle the keys -> we will marginalize the keys one by one in this order
  std::shuffle(key_list.begin(), key_list.end(), std::default_random_engine(1234));

  for (const auto& key : key_list) {
    gtsam::KeySet marginal_keys;
    marginal_keys.insert(key);
    result = updateAndMarginalize({}, {}, marginal_keys, isam);
    estimate = isam.calculateBestEstimate();
  }
}

int main(int argc, char** argv)
{
  testing::InitGoogleTest(&argc, argv);
  return RUN_ALL_TESTS();
}
```

dellaert, 2022-02-15:
@gradyrw , as it happens we were looking at marginalization today, but it's been a while since I did a deep-dive - and the students that helped create this are all gone. I am prepared to take a deeper look if you could create a some unit tests that fail. I just pushed a branch `fix/iSAM2` which you could create a PR against from a fork, if you're willing.

gradyrw, 2022-02-15:
> @gradyrw , as it happens we were looking at marginalization today, but it's been a while since I did a deep-dive - and the students that helped create this are all gone. I am prepared to take a [...]

Sounds good. I'll put up a PR with the failing tests soon.

gradyrw, 2022-04-14:
I've had a chance to look into this again and have an update. It looks like issue #3 that I brought up isn't actually a problem, there's a bug in the `markAffectedKeys` function that I was using to re-organize the graph which was causing that problem to appear. This is good since the first two issues are really easy to fix. I did find one additional issue,  `marginalizeLeaves` ignores the `findUnusedFactorSlots` parameter, however this is also relatively easy to fix.

I'll try to make a PR with all of these fixes together in the next few days.

dellaert, 2022-04-14:
great!

gradyrw, 2022-04-25:
> @gradyrw I've also ran into the situation you mentioned in issue 3. I end up with a clique `P( v90 b90 x90 x88 v88 b88 | b206 v206 x206 )` where `b88 v88 x88` are supposed to be marginalized out, but due to the ordering assumption they are not removed. You mentioned that you made your proposed solution work for the case where all variables have the same size, would you be willing to share the respective lines?

The problem turned out to be my usage of "markAffectedKeys". That function wasn't marking all of the necessary variables in order to enforce the ordering requirement. It marks all of the keys in subtrees, but misses keys belonging to the same clique as the marginal variables. In your case, the keys `{v90, b90, 90}` need to get marked, but won't be. Modifying "markAffectedKeys" so that it handles this case effectively fixes the issue. I think that I actually didn't quite do this correctly in the test code which does this in #1172

patripfr, 2022-04-25:
@gradyrw I just tested the most recent `fix/iSAM2` with your changes merged and the issue does not appear anymore. Thanks!

dellaert, 2022-04-25:
Yay!

ProfFan, 2023-07-30:
@dellaert Appears that we have not merged in the `fix/iSAM2` branch, I'll open a PR

changxi188, 2023-09-13:
We have encountered the same Issue 3, but it has not been fixed in Release 4.2. When can we expect this issue to be merged into Release 4.2

truher, 2024-08-15:
should this fix be available in the 4.2 release now?  i'm seeing issues that might be related, see https://groups.google.com/g/gtsam-users/c/qAuXDKyR1ac/m/2k_M7iTjAAAJ

ProfFan, 2024-08-16:
It's probably not, let me see what I can do

truher, 2024-08-16:
ok.  since i can work around the particular issues i'm seeing, it's not urgent.

ProfFan, 2024-08-16:
#1796