borglab/gtsam #1106: ISAM2 3D Case for g2o file

WANGYINGYU, 2022-02-16:
Hi,

I now need to use ISAM2 for some 3d datasets in g2o-like format such as sphere2500, but I did not find similar examples in the examples file, which either generated several simulated data using poses and landmarks directly or read the g2o format file and used it directly in some batch methods. I tried to write the code myself but I found it difficult because I was not familiar with the library and the format of the data.

Is there some example code for this kind of reference? Thanks.

WANGYINGYU, 2022-02-17:
> are you sure you can't find any?
I think so. VisualISAM2Example not load and use g2o file, SolverComparer uses 2d dataset. I not find the suitable code for my case. Do you have some suitable code for my case? Thank you. @dellaert

dellaert, 2022-02-17:
TEST(dataSet, readG2o3D) {
...
}

WANGYINGYU, 2022-02-22:
> 

I use readG2o(dataSet,is3D), but this api return [initial value, graph], which is good for batch optimization but it doesn't work well for incremental method (ISAM2), do you have any suggestions? Thank you. @dellaert

dellaert, 2022-02-22:
Why would it not be good for incremental? You can feed the factors *and* values in an incremental way to iSAM, after loading. `VariableIndex` might help.

I'll close this issue unless the above does not solve your problem. If you'd like a different API, feel free to contribute with unit-tested PR.

closed 2022-02-22 by dellaert

WANGYINGYU, 2022-02-23:
> Why would it not be good for incremental? You can feed the factors _and_ values in an incremental way to iSAM, after loading. `VariableIndex` might help.
> 
> I'll close this issue unless the above does not solve your problem. If you'd like a different API, feel free to contribute with unit-tested PR.

The problem is that I don't know how to use the variable index to enter values in incremental for the unique format. I used python wrapper and after using loadg2o, the graph is NonLinearFactorGraph size 4949. How do I access one of the factor  by index value? Can you give me an example? Thank you. @dellaert