borglab/gtsam #1114: The role of the `integration covariance`

goldenhazard, 2022-02-21:
Hello, I am recently navigating through `imu_preintegration factor`
While scrutinizing `ImuFactor` and `CombinedImuFactor` , I bumped into the presence of so-called `integration_covariance` or `biasAccOmegaInt`. The two terms appear while propagating uncertainty as follows,

`ImuFactor.cpp :: integrateMeasurement`
```
  const Matrix3& iCov = p().integrationCovariance;
  preintMeasCov_ = A * preintMeasCov_ * A.transpose();
  preintMeasCov_.noalias() += B * (aCov / dt) * B.transpose();
  preintMeasCov_.noalias() += C * (wCov / dt) * C.transpose();

  // NOTE(frank): (Gi*dt)*(C/dt)*(Gi'*dt), with Gi << Z_3x3, I_3x3, Z_3x3
  preintMeasCov_.block<3, 3>(3, 3).noalias() += iCov * dt;
```
`CombinedImuFactor.cpp :: integrateMeasurement`
```
  // BLOCK DIAGONAL TERMS
  D_t_t(&G_measCov_Gt) = dt * iCov;
  D_v_v(&G_measCov_Gt) = (1 / dt) * vel_H_biasAcc
      * (aCov + p().biasAccOmegaInt.block<3, 3>(0, 0))
      * (vel_H_biasAcc.transpose());
  D_R_R(&G_measCov_Gt) = (1 / dt) * theta_H_biasOmega
      * (wCov + p().biasAccOmegaInt.block<3, 3>(3, 3))
      * (theta_H_biasOmega.transpose());
  D_a_a(&G_measCov_Gt) = dt * p().biasAccCovariance;
  D_g_g(&G_measCov_Gt) = dt * p().biasOmegaCovariance;`
```

At first, I thought this was designed in heuristic to preclude some numerical failure while integrating measurement covariance. But found out it is the value that can be analytically derived, referring to your comment `G*CovMeas*GT`.

What is the role of it and how can I use it?

varunagrawal, 2022-02-22:
I am writing this up actually. Refer to #879.

goldenhazard, 2022-02-26:
Oh looking forward to the release of the docs. Thank you for your answer

varunagrawal, 2022-08-03:
@goldenhazard #874 has the full documentation of the integration covariance.

closed 2022-08-21 by varunagrawal