borglab/gtsam #1118: Error in building MATLAB wrapper: No module named 'gtwrap'

SarveshS325, 2022-02-22:
## Description

I am trying to build the MATLAB wrapper. Getting the following error
```
Traceback (most recent call last):
  File "/home/sarvesh/Libraries/gtsam/wrap/scripts/matlab_wrap.py", line 11, in <module>
    from gtwrap.matlab_wrapper import MatlabWrapper
ModuleNotFoundError: No module named 'gtwrap'
matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:86: recipe for target 'wrap/gtsam/gtsam_wrapper.cpp' failed
make[2]: *** [wrap/gtsam/gtsam_wrapper.cpp] Error 1
CMakeFiles/Makefile2:30583: recipe for target 'matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all' failed
make[1]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
Makefile:165: recipe for target 'all' failed
make: *** [all] Error 2
```

## Steps to reproduce

1. ```
cmake -DCMAKE_BUILD_TYPE=Release -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=$HOME/matlab/toolbox  -DGTSAM_BUILD_UNSTABLE:OPTION=ON  -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..

```
2. `make`

<!-- If you have a code sample, unit test, error messages, stack traces, etc., please provide it here as well -->

## Environment
Ubuntu 18.04
gtsam 4.1.1
MATLAB 2018a
Python 3.8.12 (conda env)

<!-- Please provide information about your code environment, things such as OS, language of use (C++, Matlab or Python), version of dependent libraries if using a custom build etc., anything to provide us more information. -->

<!-- The more the information, the faster we can help resolve the issue -->

## Additional information
I am able to build the python wrapper just fine
<!-- Add any other infor or context about the problem here. -->

varunagrawal, 2022-02-22:
Thanks for reporting! It'll be about a few days before I can look into this (since we're all trying to make IROS 22 deadlines), but it looks like it should be a simple fix.

SarveshS325, 2022-03-16:
Any idea where the problem could be arising from?

frank-Regal, 2022-03-17:
also getting the same error when building, any resolution?

dellaert, 2022-03-18:
@varunagrawal we don’t’ have to “git clone” with some sub tree flag, right? At least, I remember it works out of the box for me.

varunagrawal, 2022-03-18:
@SarveshS325 the issue is that you're using a conda environment, so you'll need to specify the values for `GTSAM_PYTHON_VERSION` and `PYTHON_EXECUTABLE` in CMake (using ccmake for example).

SarveshS325, 2022-03-18:
@varunagrawal, if that were the case I should not have been able to build python bindings either. I have tried with system python3 outside conda environment and setting the `GTSAM_PYTHON_VERSION` variable, same result.

varunagrawal, 2022-03-18:
Fair point. Can you please share the GTSAM branch and commit you are trying to build?

SarveshS325, 2022-03-18:
I am building from the source archive of the 4.1.1 release

varunagrawal, 2022-03-18:
Oh then it seems like a bug creeped in that was subsequently fixed. We don't have a CI for Matlab which is why it wasn't caught before, so I apologize for that.

Do you mind building the latest `develop` branch? It should have the fixes and there should be no impact to your existing code.

varunagrawal, 2022-03-18:
Quick update: even develop has some issues :-( I'll put in a PR in a few minutes.

SarveshS325, 2022-03-18:
Yup, just checked

SarveshS325, 2022-03-22:
@varunagrawal No, still the same issue

SarveshS325, 2022-03-22:
It is the same as in the original issue
```
Traceback (most recent call last):
  File "/home/sarvesh/Libraries/gtsam/wrap/scripts/matlab_wrap.py", line 11, in <module>
    from gtwrap.matlab_wrapper import MatlabWrapper
ModuleNotFoundError: No module named 'gtwrap'
matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:86: recipe for target 'wrap/gtsam/gtsam_wrapper.cpp' failed
make[2]: *** [wrap/gtsam/gtsam_wrapper.cpp] Error 1
CMakeFiles/Makefile2:30583: recipe for target 'matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all' failed
make[1]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
Makefile:165: recipe for target 'all' failed
make: *** [all] Error 2
```

varunagrawal, 2022-03-22:
Can you change that line in `matlab_wrap.py` to `from gtwrap.matlab_wrapper.wrapper import MatlabWrapper`?

varunagrawal, 2022-03-22:
This is very weird, especially since the python wrapper works for you. Both wrappers have a line `from gtwrap.X import Y` so if your system is able to build the python wrapper, then it should be able to build the Matlab wrapper.

Would you mind clearing your build directory and recompiling from scratch? The CMake cache may not have updated. Also please make sure you're on commit d0a1852c0.

SarveshS325, 2022-03-23:
I have tried starting with a squeaky clean build and I am on `d0a1852`. Maybe there is a difference between how `pybind_wrap.py` and ``

varunagrawal, 2022-06-26:
@SarveshS325 @wxchencn I recently made some updates to the wrapper. Can you please try again?

varunagrawal, 2022-06-27:
Can you try the latest develop branch please?

YeXiaoYu539, 2022-06-28:
> Can you try the latest develop branch please?

I have tried the latest develop branch and have the same error.
`  File "/home/yexiaoyu/gtsam/wrap/scripts/matlab_wrap.py", line 11, in <module>
    from gtwrap.matlab_wrapper import MatlabWrapper
ModuleNotFoundError: No module named 'gtwrap'
matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:75: recipe for target 'wrap/gtsam/gtsam_wrapper.cpp' failed
`

Thanks very much!

YeXiaoYu539, 2022-07-03:
> Can you try the latest develop branch please?

I try the version **GTSAM 4.0.3 Release** , and the problem does not appear.

Thanks very much!

- Ubuntu18.04
-  GTSAM 4.0.3
-  MATLAB R2018a
-  Python 3.6.9

varunagrawal, 2022-07-04:
Can you add this on line 254 of MatlabWrap.cmake and give us the output?

```cmake
message(">>>>>>>>>>> PYTHONPATH=${GTWRAP_PACKAGE_DIR}${GTWRAP_PATH_SEPARATOR}$ENV{PYTHONPATH}")
```

varunagrawal, 2022-07-04:
Also please paste the full message here please, starting from the output of CMake.

varunagrawal, 2022-07-04:
@SarveshS325 @YeXiaoYu539 please give us the full message and error log so we can see what is going on in your settings. We have an idea of what the issue may be but we need to confirm it.

varunagrawal, 2022-07-04:
Can you also provide your CMake versions?

YeXiaoYu539, 2022-07-04:
> Can you add this on line 254 of MatlabWrap.cmake and give us the output?
> 
> ```cmake
> message(">>>>>>>>>>> PYTHONPATH=${GTWRAP_PACKAGE_DIR}${GTWRAP_PATH_SEPARATOR}$ENV{PYTHONPATH}")
> ```

- Ubuntu18.04
- GTSAM 4.0.3
- MATLAB R2018a
- Python 3.6.9
- CMake version : 3.10.2

**Cmake Output**
```
~/gtsam-4.0.3/build
$ cmake -DEXAMPLES_BUILD_MATLAB_TOOLBOX:OPTION=ON -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=/path/install/toolbox ..
-- GTSAM_SOURCE_ROOT_DIR: [/home/yexiaoyu/gtsam-4.0.3]
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Found Eigen version: 3.3.7
-- Building 3rdparty
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building smart
-- Building navigation
-- GTSAM Version: 4.0.3
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building wrap module gtsam
-- Installing Matlab Toolbox to /path/install/toolbox
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.0.3
-- Install prefix: /usr/local
-- Building wrap module gtsam_unstable
-- Installing Matlab Toolbox to /path/install/toolbox
-- Wrote /home/yexiaoyu/gtsam-4.0.3/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--   CMAKE_CXX_COMPILER_ID type     : GNU
--   CMAKE_CXX_COMPILER_VERSION     : 7.5.0
--   CMake version                  : 3.10.2
--   CMake generator                : Unix Makefiles
--   CMake build tool               : /usr/bin/make
-- Build flags                                               
--   Build Tests                    : Enabled
--   Build examples with 'make all' : Enabled
--   Build timing scripts with 'make all': Disabled
--   Build shared GTSAM libraries   : Enabled
--   Put build type in library name : Enabled
--   Build libgtsam_unstable        : Enabled
[... 44 lines cut ...]
--   Quaternions as default Rot3     : Disabled
--   Runtime consistency checking    : Disabled
--   Rot3 retract is full ExpMap     : Enabled
--   Pose3 retract is full ExpMap    : Enabled
--   Deprecated in GTSAM 4 allowed   : Enabled
--   Point3 is typedef to Vector3    : Disabled
--   Metis-based Nested Dissection   : Enabled
--   Use tangent-space preintegration: Enabled
--   Build Wrap                      : Enabled
-- MATLAB toolbox flags                                      
--   Install MATLAB toolbox          : Enabled
--   MATLAB root                     : /home/yexiaoyu/gtsam-4.0.3/build/matlab
--   MEX binary                      : /home/yexiaoyu/gtsam-4.0.3/build/matlab/bin/mex
-- Cython toolbox flags                                      
--   Install Cython toolbox          : Disabled
-- ===============================================================
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    EXAMPLES_BUILD_MATLAB_TOOLBOX

-- Build files have been written to: /home/yexiaoyu/gtsam-4.0.3/build
```

YeXiaoYu539, 2022-07-04:
> @SarveshS325 @YeXiaoYu539 please give us the full message and error log so we can see what is going on in your settings. We have an idea of what the issue may be but we need to confirm it.

- Ubuntu18.04
- GTSAM 4.2a7
- MATLAB R2018a
- Python 3.6.9
- CMake version : 3.10.2

```
/gtsam/build$ cmake -DEXAMPLES_BUILD_MATLAB_TOOLBOX:OPTION=ON -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=/path/install/toolbox ..
-- GTSAM Version: 4.2a7
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Eigen version: 3.3.7
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.2.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.2.0
-- Install prefix: /usr/local
CMake Warning at wrap/cmake/GtwrapUtils.cmake:96 (find_package):
  By not providing "FindPython.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python" (requested
  version 3.6.9) with any of the following names:

    PythonConfig.cmake
    python-config.cmake

  Add the installation prefix of "Python" to CMAKE_PREFIX_PATH or set
  "Python_DIR" to a directory containing one of the above files.  If "Python"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  wrap/CMakeLists.txt:10 (gtwrap_get_python_version)

-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Building wrap module gtsam
-- Installing Matlab Toolbox to /path/install/toolbox
-- GTSAM_SOURCE_ROOT_DIR: [/home/yexiaoyu/gtsam]
[... 44 lines cut ...]
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.1           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Enabled
--  MATLAB root                                      : /XXX/Matlab
--  MEX binary                                       : /XXX/Matlab/bin/mex
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yexiaoyu/gtsam/build

```

varunagrawal, 2022-07-04:
Thanks for the info. Can you try fixing that Cmake warning? That may actually be your issue (and we should make that an error rather than a warning).

varunagrawal, 2022-07-04:
If you're using the system python, the fix is to do `apt install python3-dev`

varunagrawal, 2022-07-04:
@SarveshS325 @YeXiaoYu539 @frank-Regal I figured out the issue!!! PR #1235 fixes this. :tada:

closed 2022-07-05 by varunagrawal

varunagrawal, 2022-07-05:
Can you please comment on whether the latest develop branch works for you?

YeXiaoYu539, 2022-07-06:
> Can you please comment on whether the latest develop branch works for you?

Thanks very much! I have already tried the latest develop branch, it indeed fixes the problem!

PR #1140 (2022-03-21): Fix Matlab Wrapper
This PR fixes issues with the matlab wrapper and adds some documentation fixes.
Fixes #1118
PR #1235 (2022-07-05): Wrapper Update
Fixes #1118 

ca357ccdd Merge pull request #149 from borglab/install-package
886846724 set the GTWRAP_PATH_SEPARATOR properly for MatlabWrap
4abed7fa0 install the python package explicitly

git-subtree-dir: wrap
git-subtree-split: ca357ccdd27f0661e73ff7a1771768dc4bf8f746