borglab/gtsam #1119: How to obtain factorindex in ISAM2?

vigorbird, 2022-02-23:
I am using ISAM2 to optimize robot pose using GPS and VO data. But I find that the time cost is increasing with the increase of data.
I look into details of GTSAM source code and find that the ISAM2 provides another "update" function to remove unused factor in graph:
ISAM2Result ISAM2::update(
    const NonlinearFactorGraph& newFactors, const Values& newTheta,
    const FactorIndices& removeFactorIndices,
    const boost::optional<FastMap<Key, int> >& constrainedKeys,
    const boost::optional<FastList<Key> >& noRelinKeys,
    const boost::optional<FastList<Key> >& extraReelimKeys,
    bool force_relinearize) 
I want to input removeFactorIndices but I do not know how to obtain factorIndex from graph.

roger-, 2022-03-12:
Did you ever figure this out?

Seems like the factor indices are returned in [`ISAM2Result::newFactorsIndices`](https://gtsam.org/doxygen/a04364.html#adb82c057d927f3c326ca3c38dac29660).

varunagrawal, 2022-04-02:
Please ask this question on the Google Group since you will get a faster response there. If this is a legit code issue, please feel free to reopen this.

closed 2022-04-02 by varunagrawal