borglab/gtsam #1133: ISAM2Result.newFactorsIndices() missing in Python?

roger-, 2022-03-12:
I'm trying to remove some factors from ISAM2 (similar to #1119), and it seems that I need to track their indices using `gtsame.ISAM2Result.newFactorsIndices`.

Unfortunately this seems to be missing in Python:

```python
> isam = gtsam.ISAM2()
...
> res = isam.update(graph, init_vals)
> type(res)
<class 'gtsam.gtsam.ISAM2Result'>
> print(res.newFactorsIndices)
AttributeError: 'gtsam.gtsam.ISAM2Result' object has no attribute 'newFactorsIndices'
```

Is there a workaround or am I doing something wrong?

Setup: gtsam 4.1.1 / Ubuntu 20.04.4 / Python 3.8

roger-, 2022-03-14:
That's good idea, thanks! Seems like I'll have to keep track of the entire graph instead of just handing it off to ISAM2, but should work for now.

Seems like it would be easier if factors had keys, like variables do. Wonder why indices are used?

hovren, 2022-03-14:
I accessed the factors using `isam.getFactorsUnsafe()`, so there should be no reason to make a full graph yourself.

roger-, 2022-03-14:
Ah, missed that method. Thanks!

varunagrawal, 2022-03-25:
PR #1146 should add the desired method as `getNewFactorsIndices()`.

PR #1146 (2022-03-25): Wrapping Updates
1. Wrapped `measured` methods. Fixes #1142 
2. Added new method `getNewFactorsIndices()` to `ISAM2Result` and wrapped it. Fixes #1133
closed 2022-03-25 by varunagrawal