borglab/gtsam #1163: More basis wrapper

dellaert, 2022-04-09:
## Feature

Expand the usefulness of the wrapper for `basis` module.

## Motivation

`basis.i` is missing many things:
- `FitBasis` cannot take an interval [a,b]
- For BasisFactors, only M=3,4,12 is supported
- For BasisFactors, only `Chebyshev2` is supported 

## Pitch

- fix FitBasis to take [a,b] and expose in wrapper
- make BasisFactors *dynamic*, if that is possible, then all M are supported
- add templates for all currently implemented bases

varunagrawal, 2023-06-10:
@dellaert to make `BasisFactors` dynamic, I would have to change `ParameterMatrix` to not be templated. When we discussed this design choice, we chose to make ParameterMatrix templated so we would get the compile time checking, but this comes at the cost of flexibility in the wrapper.

dellaert, 2023-06-13:
I'm ok with dynamic