borglab/gtsam #1173: serialization fails for gtsam_unstable factors

shteren1, 2022-04-18:
## Description

Calling .serialize() on a gtsam_unstable factor in python wrapper fails with the following traceback:
```
Traceback (most recent call last):
  File ".../lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3444, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-16-8865b892fa88>", line 1, in <module>
    factor.serialize()
RuntimeError: unregistered class - derived class not registered or exported
```

## Steps to reproduce

1. Use the following python code snippet:
```
import gtsam
import gtsam_unstable
factor = gtsam_unstable.ProjectionFactorPPPCal3_S2(np.ones(2), gtsam.noiseModel.Isotropic.Sigmas(np.ones(2)), 0, 1, 2, gtsam.Cal3_S2())
factor.serialize()
```

## Expected behavior

.serialize() should return a string that can be used to deserialize later

## Environment

Latest pip package 4.2a6 on ubuntu 18.04

## Additional information

I tested several other factors in gtsam_unstable, they all fail to serialize.

ProfFan, 2022-04-19:
Strange, I tested them all and they work (at least if I build the Python module with CMake locally).
Is this with the PyPI package?

shteren1, 2022-04-22:
@ProfFan yes, this issue occurs with the PyPi package.

shteren1, 2022-07-08:
@varunagrawal still getting the same problem with 4.2a7.
i tried with both a clean python venv and conda environment, using ubuntu 18.04.

ProfFan, 2022-07-08:
@shteren1 This is very strange. I think you may have a global GTSAM install that is interfering with the PyPI one. I would suggest you first test in a CLEAN Ubuntu VM and report the result.

shteren1, 2022-07-08:
@ProfFan i just tried in a clean empty ubuntu 18.04 docker image and i get the same, using this docker file:

`FROM ubuntu:bionic

# Install pip
RUN apt-get update
RUN apt-get install -y python3-pip`

and then docker run -it and ran:
python3 -m pip install -U pip
python3 -m pip install gtsam==4.2a7

then entered python and ran the script in the issue description.

varunagrawal, 2022-07-14:
Hi @shteren1. I tried running a docker container with your instructions and pip can't seem to find `4.2a7`. @ProfFan do we not support Python 3.6?

shteren1, 2022-07-14:
@varunagrawal you need to update the pip version, the default old pip can't find it:
python3 -m pip install -U pip

varunagrawal, 2022-07-14:
Yup I did that, but no dice.

shteren1, 2022-07-14:
@varunagrawal here is my log from the docker session if its any help:

```
root@25a4260852e4:/# python3 -m pip install -U pip
Collecting pip
  Downloading https://files.pythonhosted.org/packages/a4/6d/6463d49a933f547439d6b5b98b46af8742cc03ae83543e4d7688c2420f8b/pip-21.3.1-py3-none-any.whl (1.7MB)
    100% |################################| 1.7MB 1.0MB/s
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed pip-21.3.1
root@25a4260852e4:/# python3 -m pip install gtsam==4.2a7
Collecting gtsam==4.2a7
  Downloading gtsam-4.2a7-cp36-cp36m-manylinux2014_x86_64.whl (21.2 MB)
     |################################| 21.2 MB 1.1 MB/s
Collecting pyparsing>=2.4.2
  Downloading pyparsing-3.0.9-py3-none-any.whl (98 kB)
     |################################| 98 kB 6.0 MB/s
Collecting numpy>=1.11.0
  Downloading numpy-1.19.5-cp36-cp36m-manylinux2010_x86_64.whl (14.8 MB)
     |################################| 14.8 MB 17.8 MB/s
Installing collected packages: pyparsing, numpy, gtsam
Successfully installed gtsam-4.2a7 numpy-1.19.5 pyparsing-3.0.9
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@25a4260852e4:/# python3
Python 3.6.9 (default, Mar 15 2022, 13:55:28)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gtsam
>>> import numpy as np
>>> import gtsam_unstable
>>> factor = gtsam_unstable.ProjectionFactorPPPCal3_S2(np.ones(2), gtsam.noiseModel.Isotropic.Sigmas(np.ones(2)), 0, 1, 2, gtsam.Cal3_S2())
>>> factor.serialize()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: unregistered class - derived class not registered or exported
>>>
root@25a4260852e4:/# exit
root@25a4260852e4:/# exit

```

varunagrawal, 2022-07-15:
I was running the docker container on a Mac and was seeing issues, but running on a Linux desktop seems to work fine. Maybe an issue with M1 chip compatibility? @ProfFan

varunagrawal, 2022-07-15:
Yes! I am able to reproduce it. Thanks for the patience @shteren1, will hopefully have a fix for you soon.

varunagrawal, 2022-07-15:
FYI this is also an issue on Ubuntu 20.04. I am using this version since CMake is able to find Python3 more easily here.

varunagrawal, 2022-07-15:
I wrote up a simple unit test

```cpp
typedef gtsam::ProjectionFactorPPP<gtsam::Pose3, gtsam::Point3, gtsam::Cal3_S2> gtsamProjectionFactorPPPgtsamPose3gtsamPoint3gtsamCal3_S2;
BOOST_CLASS_EXPORT(gtsamProjectionFactorPPPgtsamPose3gtsamPoint3gtsamCal3_S2)

TEST(ProjectionFactorPPP, Serialize) {
  Key poseKey(X(1));
  Key transformKey(T(1));
  Key pointKey(L(1));

  Point2 measurement(323.0, 240.0);

  gtsamProjectionFactorPPPgtsamPose3gtsamPoint3gtsamCal3_S2 factor(measurement, model, poseKey, transformKey, pointKey, K);

  // Serialize the factor
  std::string serialized = serialize(factor);

//  // And de-serialize it
//  TestProjectionFactor deserializedFactor;
//  deserializeXML(serialized, deserializedFactor);
}
```

and it throws the same exception so it looks like factor serialization is broken.