borglab/gtsam #1174: Drafting 4.2 Release

ProfFan, 2022-04-19:
Since now that the annoying TBB bug has been finally resolved (#1158) I think it's probably a good time for a 4.2 release.

Outstanding issues and PRs:
- #1173 
- #292 
- #368
- #1129 
- #1122 
- #1101 
- #1098 
- #846

@varunagrawal @jlblancoc @gchenfc @dellaert @ayushbaid @acxz and all,
feel free to add the issues and PRs you think are important to this list :)

dellaert, 2022-04-19:
We are finalizing the book draft v1.0 by the end of the semester, so I might need one more alpha release to make sure everything works there.

I will work on #1098 as well.

jlblancoc, 2022-04-20:
Now that you are considering API break, would it be a good time to merge these two commits, which fixed commonly-found crashes? 
- https://github.com/borglab/gtsam/commit/a29b4dbd7418b6be4748023fd381648b59872314
- https://github.com/borglab/gtsam/commit/07bace5e057d305674b0b17c47a2bc56bb7926a9

(cc: as mentioned https://github.com/borglab/gtsam/issues/292#issuecomment-1009977935 )

@ProfFan You could add https://github.com/borglab/gtsam/pull/1175 to the list, I think it's fairly straightforward and will fix the linked side project.

It would be great to make progress on https://github.com/borglab/gtsam/issues/292 too, but at present I lack spare time for this, sorry...

So good to see the big changes that 4.2 is bringing, btw! :-)

dellaert, 2022-04-20:
@jlblancoc yeah, feel free to add a small PR with these two commits, although I'd love to have some discussion on native behavior in PR, if only to document.

varunagrawal, 2023-12-30:
Closing this since 4.2 has been released.

closed 2023-12-30 by varunagrawal