borglab/gtsam #1178: Python Wrapper TypeError

troyvicsik27, 2022-04-23:
Hello,

My teammates and I are currently trying to get the Python wrapper up and running on our computers, but we keep encountering a TypeError. I am using WSL, and my teammate is using Mac; however, the error did not appear when he tried running in Linux instead. The error is with line 157 in classes.py, and is as follows:

in
self.ctype = ctype[0] # ParseResult is a list
TypeError: 'Type' object is not subscriptable

The full error message that pops up when trying to use make python-install can be seen below:

[image]
[image]

Any idea what's going on or how to resolve this? Thank you very much!

varunagrawal, 2022-04-30:
Are you doing some custom parsing?
Also, can you please post the result of `pip show pyparsing`?

Essentially, the issue is that when `pyparsing` is reading the .i file, it should return a list as the result of parsing a class definition, but it seems like in your case, it is returning just the object directly.

varunagrawal, 2022-05-25:
Closing due to lack of response. Please reopen if this is still an issue for you.

closed 2022-05-25 by varunagrawal