borglab/gtsam #1184: Error in building MATLAB wrapper:   TypeError: unhashable type: 'Typename'

xiaocaiyaa, 2022-05-02:
I am trying to build the MATLAB wrapper. Getting the following error

...
[ 94%] Built target SmartProjectionFactorExample
[ 94%] Built target FixedLagSmootherExample
[ 97%] Built target SmartRangeExample_plaza1
[ 97%] Generating ../wrap/gtsam/gtsam_wrapper.cpp
[MatlabWrapper] Ignoring classes: ['gtsam::Point2', 'gtsam::Point3', 'gtsam::CustomFactor']
Traceback (most recent call last):
  File "/home/air/gtsam-develop/wrap/scripts/matlab_wrap.py", line 58, in <module>
    cc_content = wrapper.wrap(sources, path=args.out)
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 1639, in wrap
    self.wrap_namespace(module)
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 1034, in wrap_namespace
    self.wrap_namespace(element)
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 1040, in wrap_namespace
    class_text = self.wrap_instantiated_class(
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 964, in wrap_instantiated_class
    self.wrap_class_constructors(
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 660, in wrap_class_constructors
    varargin=self._wrap_variable_arguments(
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 237, in _wrap_variable_arguments
    check_type = self._format_type_name(
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/mixins.py", line 134, in _format_type_name
    self._format_type_name(type_name.instantiations[idx],
  File "/home/air/gtsam-develop/wrap/gtwrap/matlab_wrapper/mixins.py", line 112, in _format_type_name
    formatted_type_name += self.data_type.get(name) or name
TypeError: unhashable type: 'Typename'
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:76：wrap/gtsam/gtsam_wrapper.cpp] error 1
make[1]: *** [CMakeFiles/Makefile2:32161：matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] error 2
make: *** [Makefile:163：all] error 2

Could you help me to find what's wrong with it?
Python 3.8.10
Ubuntu 20.04.1 LTS
GTSAM 4.1.1

richardXia7462, 2022-05-08:
I also meet this problem

water137780117, 2022-05-15:
The same issue. macOS 12.3.1 (M1).

carlatcrown, 2022-05-25:
Same issue
Ubuntu Focal 20.04
MatLab 2020b

Works with 4.2a5, fails with 4.2a6.

cmake: CMAKE_ARGS=-DGTSAM_USE_SYSTEM_EIGEN:BOOLEAN=FALSE -DGTSAM_WITH_EIGEN_MKL:BOOLEAN=FALSE -DGTSAM_WITH_TBB:BOOLEAN=FALSE -DGTSAM_BUILD_EXAMPLES_ALWAYS:BOOL=FALSE -DGTSAM_BUILD_UNSTABLE:OPTION=OFF -DCMAKE_BUILD_TYPE:STRING=Release -DGTSAM_BUILD_WITH_MARCH_NATIVE:BOOL=OFF -DGTSAM_INSTALL_MATLAB_TOOLBOX:BOOLEAN=true -DMEX_COMMAND:STRING=/usr/local/MATLAB/R2020b/bin/mex
CXX_FLAGS=-DEIGEN_DONT_VECTORIZE -DEIGEN_DISABLE_UNALIGNED_ARRAY_ASSERT  -march=skylake

varunagrawal, 2022-05-25:
Sorry for the late response. I was able to reproduce and fix the issue. Once #1205 gets merged, you can just get the latest `develop` and everything should work.

If you still face any problems, please feel free to open a new issue.

carlatcrown, 2022-05-26:
Thanks for the fix - confirmed working.

PR #1205 (2022-06-02): Fix templated type declaration in geometry.i
For using templated types in methods in the wrapper, we need to use `This` instead of `Class<T>`.

This may be a strong requirement, but it is convenient and unless there is a strong argument for changing this default behavior, I'm going to keep it as is.

Fixes #1184
closed 2022-06-02 by varunagrawal