borglab/gtsam #1186: error: invalid initialization of reference of type ‘const std::vector<bool>&’ from expression of type ‘const std::vector<long unsigned int>’

ZJYone265021, 2022-05-05:
## Feature

## Motivation

## Pitch

## Alternatives

## Additional context

zkk-slam, 2022-06-08:
You can try lowering the version of gcc and g++

closed 2022-06-08 by varunagrawal

hfr2015, 2022-11-01:
> Sorry, but the error is clear. You're trying to initialize a vector of type bool with type long unsigned int. The bit storage for the two are incompatible (1 vs 4 bits) hence this error. Plus you've not provided any details on this issue making it impossible for us to debug it. If you still feel this is a valid bug, please include all the details as specified in the issue template.

tag: wrap-export
file: gtsam/gtsam_unstable/slam/PartialPriorFactor.h:124:53: error: invalid initialization of reference of type ‘const std::vector<bool>&’ from expression of type ‘const std::vector<long unsigned int>’