borglab/gtsam #1188: Using ceres and GTSAM at the same time, the program is stuck in the update function of the gtsam optimizer and cannot jump out.

Sonne-Zhu, 2022-05-05:
## Description

Hello, I implemented a SLAM algorithm based on factor graph optimization (LIO-SAM)  on ubuntu18.04. I used GTSAM4.0.0 and ceres2.0.0 in one node at the same time. After compiling, when the algorithm is executed to the update( of the isam optimizer in gtsam ) function, it will be stuck in the update function and cannot be jumped out, as if it has fallen into an infinite loop, but the program does not report an error. If ceres is not compiled, the update function of gtsam can be used normally. Do you know what is the reason for this, or how to avoid it. 

## Environment

ubuntu18.04, C++, GTSAM4.0.0, ceres2.0.0.

ProfFan, 2022-05-05:
Hello @Sonne-Zhu could you provide more information (for example, a core dump for the application when the application is stuck)? Also, please try the current develop branch and see if the problem is still there.

varunagrawal, 2023-06-16:
Closing due to lack of response. Please re-open if you still encounter an issue and please provide a reproducible example and/or more info.

closed 2023-06-16 by varunagrawal