borglab/gtsam #1189: ISAM2 update() function Error: double free or corruption (out)

dinvincible98, 2022-05-05:
## Description
When I tried to run the isam2 [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp),
I got this error: double free or corruption (out).

## Steps to reproduce

1. Build (Add -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF in cmake) and install the gtsam4.1.1
2. Run the  [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp)

Line 139 in [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp)

``` 
  // Incremental solution
  isam.update(newgraph, initialEstimate);
```

Output:

Running main() from /build/googletest-j5yxiC/googletest-1.10.0/googletest/src/gtest_main.cc
double free or corruption (out)

## Environment

Ubuntu 20.04
C++
gtsam-4.1.1

## Additional information

ProfFan, 2022-06-05:
Hi @dinvincible98 , could you run GTSAM with valgrind on the same example?

varunagrawal, 2022-07-07:
Closing due to lack of response. Please reopen if the issue still persists.

Gatsby23, 2024-05-08:
> ## Description
> When I tried to run the isam2 [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp), I got this error: double free or corruption (out).
> 
> ## Steps to reproduce
> 1. Build (Add -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF in cmake) and install the gtsam4.1.1
> 2. Run the  [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp)
> 
> Line 139 in [imu factor example code](https://github.com/borglab/gtsam/blob/develop/examples/ImuFactorsExample2.cpp)
> 
> ```
>   // Incremental solution
>   isam.update(newgraph, initialEstimate);
> ```
> 
> Output:
> 
> Running main() from /build/googletest-j5yxiC/googletest-1.10.0/googletest/src/gtest_main.cc double free or corruption (out)
> 
> GDB: (gdb) backtrace #0 __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50 #1 0x00007ffff7195859 in __GI_abort () at abort.c:79 #2 0x00007ffff720029e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff732a298 "%s\n") at ../sysdeps/posix/libc_fatal.c:155 #3 0x00007ffff720832c in malloc_printerr (str=str@entry=0x7ffff732c670 "double free or corruption (out)") at malloc.c:5347 #4 0x00007ffff7209fd0 in _int_free (av=0x7ffff735fb80 <main_arena>, p=0x5555555fbe10, have_lock=) at malloc.c:4314 #5 0x00007ffff7e27630 in gtsam::NoiseModelFactor::linearize(gtsam::Values const&) const () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #6 0x00007ffff7e289bb in gtsam::(anonymous namespace)::_LinearizeOneFactor::operator()(tbb::blocked_range const&) const () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #7 0x00007ffff7e28d7d in tbb::interface9::internal::start_for<tbb::blocked_range, gtsam::(anonymous namespace)::_LinearizeOneFactor, tbb::auto_partitioner const>::execute() () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #8 0x00007ffff7782545 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2 #9 0x00007ffff778280f in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2 #10 0x00007ffff777fb68 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2 #11 0x00007ffff7e2a9fd in gtsam::NonlinearFactorGraph::linearize(gtsam::Values const&) const () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #12 0x00007ffff7defb3d in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::ISAM2UpdateParams const&) () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #13 0x00007ffff7deeda6 in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, std::vector<unsigned long, tbb::tbb_allocator > const&, boost::optional<gtsam::FastMap<unsigned long, int> > const&, boost::optional<gtsam::FastList > const&, boost::optional<gtsam::FastList > const&, bool) () from /home/mingqingyuan/project/third_party/x86_64/lib/libgtsam.so.4 #14 0x00005555555741bc in ?? () #15 0x00005555555c23e1 in ?? () #16 0x00005555555b5856 in ?? () #17 0x00005555555b59b5 in ?? () #18 0x00005555555b5a9d in ?? () #19 0x00005555555b5fbc in ?? () #20 0x00005555555c2951 in ?? () #21 0x00005555555b61ec in ?? () #22 0x0000555555571284 in ?? () #23 0x00007ffff71970b3 in __libc_start_main (main=0x555555571240, argc=3, argv=0x7fffffffd868, init=, fini=, rtld_fini=, stack_end=0x7fffffffd858) at ../csu/libc-start.c:308 #24 0x00005555555712fe in ?? ()
> 
> ## Environment
> Ubuntu 20.04 C++ gtsam-4.1.1
> 
> ## Additional information

I have met the same problem with you, could you please give me some advices about how to solve it ?