borglab/gtsam #1192: incompatible function arguments

simin75simin, 2022-05-07:
os: ubuntu 18.04
python: 3.6.9

I was trying to run code from a [ros testing script](http://docs.ros.org/en/melodic/api/gtsam/html/test__Triangulation_8py_source.html) because I want to use the triangulation functions. However, the following occurs after many 
```
╰─$ python3 test_Triangulation.py 
setUp done
.setUp done
.setUp done
.setUp done
E
======================================================================
ERROR: test_triangulation_robust_three_poses (__main__.TestTriangulationExample)
Ensure triangulation with a robust model works.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_Triangulation.py", line 220, in test_triangulation_robust_three_poses
    model=model)
TypeError: triangulatePoint3(): incompatible function arguments. The following argument types are supported:
    1. (poses: std::vector<gtsam::Pose3, std::allocator<gtsam::Pose3> >, sharedCal: gtsam.gtsam.Cal3_S2, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    2. (poses: std::vector<gtsam::Pose3, std::allocator<gtsam::Pose3> >, sharedCal: gtsam.gtsam.Cal3DS2, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    3. (poses: std::vector<gtsam::Pose3, std::allocator<gtsam::Pose3> >, sharedCal: gtsam.gtsam.Cal3Bundler, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    4. (cameras: gtsam::CameraSet<gtsam::PinholeCamera<gtsam::Cal3_S2> >, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    5. (cameras: gtsam::CameraSet<gtsam::PinholeCamera<gtsam::Cal3Bundler> >, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    6. (cameras: gtsam::CameraSet<gtsam::PinholeCamera<gtsam::Cal3Fisheye> >, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]
    7. (cameras: gtsam::CameraSet<gtsam::PinholeCamera<gtsam::Cal3Unified> >, measurements: std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > >, rank_tol: float, optimize: bool) -> numpy.ndarray[numpy.float64[3, 1]]

Invoked with: Pose3Vector[R: [
        6.12323e-17, 6.12323e-17, 1;
        -1, 3.7494e-33, 6.12323e-17;
        -0, -1, 6.12323e-17
]
t: 0 0 1, R: [
        6.12323e-17, 6.12323e-17, 1;
        -1, 3.7494e-33, 6.12323e-17;
        0, -1, 6.12323e-17
]
t: 6.12323e-17          -1           1, R: [
        -0.198669, 0.0978434, 0.97517;
        -0.97517, 0.0795999, -0.206656;
        -0.0978434, -0.992013, 0.0795999
]
t: -0.1 -0.1    3], Cal3_S2[
        1500, 0, 640;
        0, 1200, 480;
        0, 0, 1
]
, Point2Vector[590
552, 190
432, 186.695
1074.73]; kwargs: rank_tol=1e-09, optimize=True, model=huber (1.345)
unit (2) 

Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>,
<pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic
conversions are optional and require extra headers to be included
when compiling your pybind11 module.

----------------------------------------------------------------------
Ran 4 tests in 0.001s

FAILED (errors=1)
```

varunagrawal, 2022-06-21:
Okay so the test that is failing is part of our CI system and it passes both there and on my system. This most likely indicates either something is up on your system or python 3.6.9 is being difficult.

Can you please try in a docker container and report if it fails there as well?

varunagrawal, 2022-08-03:
@simin75simin I recently pushed some fixes to the python wrapper, so can you please try with the latest `develop`?

varunagrawal, 2022-12-24:
Closing due to lack of a response. @simin75simin please reopen if the issue still persists.

closed 2022-12-24 by varunagrawal