borglab/gtsam #1200: Is it possible to specify different distributions for the noise model?

NickPerezCarletonUniversity, 2022-05-18:
when specifying a noise model for a factor via:

```noiseModel::Diagonal::Sigmas```

the noise model always assumes a Gaussian distribution. Is there a way to use different types of distributions for the noise model? Such as an [exponential distribution](https://en.wikipedia.org/wiki/Exponential_distribution).

dellaert, 2022-05-19:
Look at robust noise models. Distribution = exp(-error)

NickPerezCarletonUniversity, 2022-05-30:
Hi @dellaert thanks for the help. I am not quit sure what you mean though by 'look at robust noise models'.

I see that there are [definitions for robust noise](https://github.com/borglab/gtsam/blob/develop/gtsam/linear/NoiseModel.h#L640), however I'm not sure how to write the code for what you are suggesting.

For creating a Gaussian noise model I can simply just write:

'''auto model = noiseModel::Diagonal::Sigmas((Vector(6)<<0.003, 0.003, 0.003, 0.05, 0.05, 0.05).finished());'''

but I'm not sure how to do the same for an exponential distribution. I don't see anywhere in the code how to specify the distribution.

Some thing like this?...

'''auto model = noiseModel::Robust::Create(?...);'''

It would be nice to have an example showing how to specify a specific distribution for a noise model in the examples folder.

gchenfc, 2022-07-20:
https://github.com/borglab/gtsam/blob/develop/tests/testRobust.cpp may help:

https://github.com/borglab/gtsam/blob/0af17f438f62f4788f3a572ecd36d06d224fd1e1/tests/testRobust.cpp#L33-L36

together with https://github.com/borglab/gtsam/blob/develop/doc/robust.pdf

varunagrawal, 2023-06-05:
@catproof you should also check out Frank's book for an easy introduction to this. Closing for now. :)

closed 2023-06-05 by varunagrawal