borglab/gtsam #1211: double free or corruption (out) when using smart factor

tccoin, 2022-06-02:
## Description

I am currently using `SmartProjectionPoseFactor` with `CombinedImuFactor` in my project and sometimes the program will fail with the error `double free or corruption (out)`. By using address sanitizer, I managed to find the code which triggers the error.

[image]
[image]

## Steps to reproduce

My experience is, using `SmartProjectionPoseFactor` or  `CombinedImuFactor` alone won't trigger this error, but using them both will. I am not sure how this is related to the memory error but in case I just provide what I found.

## Expected behavior

The systems should not fail.

## Environment

Ubuntu 20.04 in docker.

## Additional information

ProfFan, 2022-06-20:
My gut feeling is that this is b/c a missing lock and the two threads are having a race condition. Don't have the bandwidth to debug this, but I feel that if you add locking to the two mutable `cameraPosesTriangulation_` it should not crash

tccoin, 2022-06-20:
Thank you! I will have a try.

closed 2022-07-15 by tccoin