borglab/gtsam #1216: Build hangs up at 8>TangentPreintegration.cpp (Windows 10)

teku45, 2022-06-03:
Hi,

I am running into an issue where when I attempt to build GTSAM on Windows 10, the build process always hangs up at the following sequence of files:

```
.
.
.
8>ImuFactor.cpp
8>ManifoldPreintegration.cpp
8>NavState.cpp
8>PreintegratedRotation.cpp
8>PreintegrationBase.cpp
8>PreintegrationParams.cpp
8>ScenarioRunner.cpp
8>TangentPreintegration.cpp
```

The project was generate with CMake 3.23.2 for windows, and I am using Visual Studio 2022

Thanks

ProfFan, 2022-06-05:
If you build GTSAM with Python wrapper turned on it is going to take a long time. But hanging is a different issue. It would be great if you can provide more verbose logs.

teku45, 2022-06-06:
Hi, 

Thank you. Disabling the python seems to have made the build take a shorter amount of time. 

Incidentally, I am also trying to build GTSAM on an Ubuntu 20.04 machine and am running into a separate issue while running make install. I am mainly trying to compile the Matlab toolbox. I have Python 3.8 as well as the pyparsing package but it seems that it is still unable to find the pyparsing package somehow. Do you have any thoughts on this? Here is the full output log from make install.

```
[  2%] Built target CppUnitLite
[ 20%] Built target metis-gtsam
[ 56%] Built target gtsam
[ 56%] Generating wrap/gtsam/gtsam_wrapper.cpp
[ 58%] Built target UGM_chain
[ 58%] Built target VisualISAMExample
[ 58%] Built target easyPoint2KalmanFilter
[ 58%] Built target SolverComparer
[ 58%] Built target StereoVOExample_large
[ 58%] Built target ShonanAveragingCLI
[ 58%] Built target StereoVOExample
Scanning dependencies of target SelfCalibrationExample
Scanning dependencies of target RangeISAMExample_plaza2
Scanning dependencies of target SFMExample
Scanning dependencies of target SFMExample_SmartFactorPCG
Scanning dependencies of target SFMExample_bal
Scanning dependencies of target LocalizationExample
Scanning dependencies of target METISOrderingExample
[ 61%] Building CXX object examples/CMakeFiles/SFMExample.dir/SFMExample.cpp.o
[ 61%] Building CXX object examples/CMakeFiles/SelfCalibrationExample.dir/SelfCalibrationExample.cpp.o
[ 64%] Building CXX object examples/CMakeFiles/METISOrderingExample.dir/METISOrderingExample.cpp.o
[ 64%] Building CXX object examples/CMakeFiles/LocalizationExample.dir/LocalizationExample.cpp.o
[ 64%] Building CXX object examples/CMakeFiles/RangeISAMExample_plaza2.dir/RangeISAMExample_plaza2.cpp.o
[ 64%] Building CXX object examples/CMakeFiles/SFMExample_bal.dir/SFMExample_bal.cpp.o
[ 64%] Building CXX object examples/CMakeFiles/SFMExample_SmartFactorPCG.dir/SFMExample_SmartFactorPCG.cpp.o
Traceback (most recent call last):
  File "/home/siddsubr/gtsam-4.1.0/wrap/matlab_wrapper.py", line 5, in <module>
    import interface_parser as parser
  File "/home/siddsubr/gtsam-4.1.0/wrap/interface_parser.py", line 14, in <module>
    from pyparsing import (
ModuleNotFoundError: No module named 'pyparsing'
make[2]: *** [gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:73: gtsam/wrap/gtsam/gtsam_wrapper.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:3153: gtsam/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 64%] Linking CXX executable METISOrderingExample
[ 64%] Linking CXX executable LocalizationExample
[ 64%] Built target METISOrderingExample
[ 64%] Built target LocalizationExample
[ 64%] Linking CXX executable SelfCalibrationExample
[ 64%] Built target SelfCalibrationExample
[ 64%] Linking CXX executable SFMExample
[ 64%] Built target SFMExample
[ 64%] Linking CXX executable SFMExample_bal
[ 64%] Built target SFMExample_bal
[ 64%] Linking CXX executable RangeISAMExample_plaza2
[ 64%] Built target RangeISAMExample_plaza2
[ 66%] Linking CXX executable SFMExample_SmartFactorPCG
[ 66%] Built target SFMExample_SmartFactorPCG
make: *** [Makefile:163: all] Error 2
```

teku45, 2022-06-07:
Resolved this by using Ubuntu 18.04 instead of 20.04

closed 2022-06-07 by teku45

laxnpander, 2022-11-23:
Hey all, 

same issue for me on 20.04. I can provide the following CMake config:

```
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- The CXX compiler identification is MSVC 19.29.30146.0
-- The C compiler identification is MSVC 19.29.30146.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM Version: 4.2a7
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
CMake Warning (dev) at cmake/HandleBoost.cmake:28 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable Boost_ROOT is set to:

    D:\Workspace\spear\spear\3rd\boost

  For compatibility, CMake is ignoring the variable.
Call Stack (most recent call first):
  CMakeLists.txt:54 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Found Boost: D:/Workspace/spear/spear/3rd/boost (found suitable version "1.72.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex atomic
-- Found Eigen version: 3.3.7
-- Looking for execinfo.h
-- Looking for execinfo.h - not found
-- Looking for getline
-- Looking for getline - not found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES)
-- Found OpenMP_C: -openmp (found version "2.0")
-- Found OpenMP_CXX: -openmp (found version "2.0")
-- Found OpenMP: TRUE (found version "2.0")
-- Could NOT find TBB (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES tbb tbbmalloc) (Required is at least version "4.4")
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS)
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- Adding precompiled header for MSVC
-- GTSAM Version: 4.2.0
-- Install prefix: D:/Workspace/spear/spear/3rd/gtsam/install
-- Building GTSAM - shared: ON
-- Wrote D:/Workspace/spear/spear/3rd/gtsam/cmake-build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
[... 33 lines cut ...]
--  Using Boost version                              : 107200
--  Use Intel TBB                                    : TBB not found
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : STL
--  Cheirality exceptions enabled                    : YES
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.1           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags
--  Build Python module with pybind                  : Disabled
-- ===============================================================
```

I don't think it crashed, but it compiles for 3h now on a rather weak laptop. Python bindings are deactivated. Will report back if it finishes the build at some point.

ds58, 2023-06-07:
In case anyone else runs into this on Windows, make sure you're using the latest commits. It's only an issue with the older releases of gtsam.

The issue is related to the built-in Eigen version in the older releases. In my testing, anything less than 3.4.0 causes the build to be way longer. If you use an older release of gtsam, you can install Eigen manually (3.4.0+) and use `-DGTSAM_USE_SYSTEM_EIGEN=ON` to resolve it.

varunagrawal, 2023-06-07:
Can confirm. I worked with @ds58 to figure this out.