borglab/gtsam #1223: isam2 keys in python can not be reset

davidezilio, 2022-06-17:
## Description

I am using isam2 (in python) and noticed that I can't reset the key value following the calculateEstimate and (the graph resize) step.

## Steps to reproduce

Simply run IMUKittyExampleGPS.py but use an additional variable for the keys which is reset after the optimization step.
See example attached.
[IMUKittiExampleGPS.zip](https://github.com/borglab/gtsam/files/8928259/IMUKittiExampleGPS.zip)

## Expected behavior
However I did saw this being reset in several github repo using gtsam isam2 in c++.
See example here
https://github.com/TixiaoShan/LIO-SAM/blob/master/src/imuPreintegration.cpp#L346

Is this an issue or am I missing something?

varunagrawal, 2022-06-20:
I guess we just haven't wrapped that functionality in C++ to Python. Should be an easy fix hopefully, unless there is some reason the wrapper doesn't support it yet.

varunagrawal, 2022-06-20:
@davidezilio I looked at your shared code, and I am not quite sure I follow what the issue is. It looks like all you need is for `i_key` to be reset to 0 when `i > (first_gps_pose + 2 * gps_skip)`?
`i_key` in this case is a Python int and doesn't have any dependence on ISAM2, so maybe elaborating a bit more to help us understand your issue please?

davidezilio, 2022-06-27:
Thanks a lot for your reply.
_"It looks like all you need is for i_key to be reset to 0 when i > (first_gps_pose + 2 * gps_skip)"_
That is correct, but when i try to do that (as done in the script in line 333) I get the following error

RuntimeError: Attempting to add a key-value pair with key "b1", key already exists.

But I have seen the key value to be reset successfully in C++ repo.

chrisflesher, 2024-11-14:
This seems to be working ok for me:

```
        isam2.update(graph, values)
        graph.resize(0)
        values.clear()
```