borglab/gtsam #1231: Cannot Progress CMake for macOS Monterey 12.4

zarathustr, 2022-07-02:
After installing dependencies via Brew on the machine with macOS Monterey 12.4, we are unable to pass the CMake (3.22) check. The messages are:
```
CMake Error at gtsam_unstable/CMakeLists.txt:1 (project):
  The CMAKE_CXX_COMPILER:

    clang++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```

Seems that the CMake cannot properly handle the ```project(gtsam LANGUAGES CXX)```. Is there any fix for this issue?

varunagrawal, 2022-07-03:
This seems to be an issue on your system. Can you verify that `clang++` is in your PATH variable? You should be able to call `clang++` from the commandline.

zarathustr, 2022-07-10:
@varunagrawal Hi Varun. Thanks for your reply. I believe that this is because of the newest CMake 3.23 which was installed automatically with Brew on macOS. I downgrade the source repo to many old-time versions but still seems to be not working. I managed to pass the CMake issues via the help of internal CMake tools of the CLion IDE. I reproduce the success via a manual compilation of CMake 3.5, which verifies this reason. Maybe we need some modifications to adapt for the new CMake toolset.

varunagrawal, 2022-12-29:
Closing. If the issue still persists, please reopen.

closed 2022-12-29 by varunagrawal