borglab/gtsam #1242: Missing geometry classes in Python wrapper (4.1.1)

btalb, 2022-07-13:
## Description

The `Point2()` and `Point3()` classes are missing from the Python library when building GTSAM 4.1.1

## Steps to reproduce

1. Clone and checkout 4.1.1:
    ```
    git clone https://github.com/borglab/gtsam; cd gtsam
    git checkout 4.1.1
    ```
2. Build with Python wrappers:
    ```
    mkdir build
    cd build
    cmake .. -DGTSAM_BUILD_PYTHON=1
    make
    ```
3. Attempt to use the build `*.so`:
    ```
    cd python/gtsam
    ipython3
    ```

Get the following error when trying to use the classes:
```
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.26.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import gtsam

In [2]: gtsam.__file__
Out[2]: '/home/ben/tmp/gtsam/build/python/gtsam/gtsam.cpython-38-x86_64-linux-gnu.so'

In [3]: gtsam.Point2
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-4dfb95120b36> in <module>
----> 1 gtsam.Point2

AttributeError: module 'gtsam' has no attribute 'Point2'
```

## Expected behavior

I would expect those classes to be available. Tracing back, the `geometry.i` interface seems to have them defined:

https://github.com/borglab/gtsam/blob/69a3a75195b65356d6e56669e5199d325c7962c9/gtsam/geometry/geometry.i#L8-L33

But they're nowhere to be found in the `geometry.cpp` file `wrap` generates for `pybind11`.

## Environment

Reproduced across machines, in manylinux Docker images, and more.

btalb, 2022-07-13:
<details>
<summary>geometry.cpp</summary>

```cpp
/**
 * @file    geometry.cpp
 * @brief   The auto-generated wrapper C++ source code.
 * @author  Duy-Nguyen Ta, Fan Jiang, Matthew Sklar, Varun Agrawal
 * @date    Aug. 18, 2020
 *
 * ** THIS FILE IS AUTO-GENERATED, DO NOT MODIFY! **
 */

// Include relevant boost libraries required by GTSAM
#include <boost/shared_ptr.hpp>

#include <pybind11/eigen.h>
#include <pybind11/stl_bind.h>
#include <pybind11/pybind11.h>
#include <pybind11/operators.h>
#include <pybind11/functional.h>
#include <pybind11/iostream.h>
#include "gtsam/config.h"
#include "gtsam/base/serialization.h"
#include "gtsam/base/utilities.h"  // for RedirectCout.

// These are the included headers listed in `gtsam.i`
#include "gtsam/geometry/Point2.h"
#include "gtsam/geometry/StereoPoint2.h"
#include "gtsam/geometry/Point3.h"
#include "gtsam/geometry/Rot2.h"
#include "gtsam/geometry/SO3.h"
#include "gtsam/geometry/SO4.h"
#include "gtsam/geometry/SOn.h"
#include "gtsam/geometry/Quaternion.h"
#include "gtsam/geometry/Rot3.h"
#include "gtsam/geometry/Pose2.h"
#include "gtsam/geometry/Pose3.h"
#include "gtsam/geometry/Unit3.h"
#include "gtsam/geometry/EssentialMatrix.h"
#include "gtsam/geometry/Cal3_S2.h"
#include "gtsam/geometry/Cal3DS2_Base.h"
#include "gtsam/geometry/Cal3DS2.h"
#include "gtsam/geometry/Cal3Unified.h"
#include "gtsam/geometry/Cal3Fisheye.h"
#include "gtsam/geometry/Cal3_S2Stereo.h"
#include "gtsam/geometry/Cal3Bundler.h"
#include "gtsam/geometry/CalibratedCamera.h"
#include "gtsam/geometry/PinholeCamera.h"
#include "gtsam/geometry/Similarity3.h"
#include "gtsam/geometry/SimpleCamera.h"
#include "gtsam/geometry/StereoCamera.h"
#include "gtsam/geometry/triangulation.h"
#include "gtsam/geometry/BearingRange.h"

#include <boost/serialization/export.hpp>

// Export classes for serialization
BOOST_CLASS_EXPORT(gtsam::StereoPoint2)
BOOST_CLASS_EXPORT(gtsam::Rot2)
BOOST_CLASS_EXPORT(gtsam::SOn)
BOOST_CLASS_EXPORT(gtsam::Rot3)
BOOST_CLASS_EXPORT(gtsam::Pose2)
BOOST_CLASS_EXPORT(gtsam::Pose3)
BOOST_CLASS_EXPORT(gtsam::Unit3)
BOOST_CLASS_EXPORT(gtsam::Cal3_S2)
BOOST_CLASS_EXPORT(gtsam::Cal3DS2_Base)
BOOST_CLASS_EXPORT(gtsam::Cal3DS2)
BOOST_CLASS_EXPORT(gtsam::Cal3Unified)
BOOST_CLASS_EXPORT(gtsam::Cal3Fisheye)
[... 815 lines cut ...]
        .def_static("Measure",[](const gtsam::Pose2& pose, const gtsam::Point2& point){return gtsam::BearingRange<gtsam::Pose2, gtsam::Point2, gtsam::Rot2, double>::Measure(pose, point);}, py::arg("pose"), py::arg("point"))
        .def_static("MeasureBearing",[](const gtsam::Pose2& pose, const gtsam::Point2& point){return gtsam::BearingRange<gtsam::Pose2, gtsam::Point2, gtsam::Rot2, double>::MeasureBearing(pose, point);}, py::arg("pose"), py::arg("point"))
        .def_static("MeasureRange",[](const gtsam::Pose2& pose, const gtsam::Point2& point){return gtsam::BearingRange<gtsam::Pose2, gtsam::Point2, gtsam::Rot2, double>::MeasureRange(pose, point);}, py::arg("pose"), py::arg("point"));

    m_.def("align",[](const gtsam::Point2Pairs& pairs){return gtsam::align(pairs);}, py::arg("pairs"));
    m_.def("triangulatePoint3",[](const gtsam::Pose3Vector& poses, boost::shared_ptr<gtsam::Cal3_S2> sharedCal, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(poses, sharedCal, measurements, rank_tol, optimize);}, py::arg("poses"), py::arg("sharedCal"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::Pose3Vector& poses, boost::shared_ptr<gtsam::Cal3DS2> sharedCal, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(poses, sharedCal, measurements, rank_tol, optimize);}, py::arg("poses"), py::arg("sharedCal"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::Pose3Vector& poses, boost::shared_ptr<gtsam::Cal3Bundler> sharedCal, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(poses, sharedCal, measurements, rank_tol, optimize);}, py::arg("poses"), py::arg("sharedCal"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::CameraSetCal3_S2& cameras, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(cameras, measurements, rank_tol, optimize);}, py::arg("cameras"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::CameraSetCal3Bundler& cameras, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(cameras, measurements, rank_tol, optimize);}, py::arg("cameras"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::CameraSetCal3Fisheye& cameras, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(cameras, measurements, rank_tol, optimize);}, py::arg("cameras"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
    m_.def("triangulatePoint3",[](const gtsam::CameraSetCal3Unified& cameras, const gtsam::Point2Vector& measurements, double rank_tol, bool optimize){return gtsam::triangulatePoint3(cameras, measurements, rank_tol, optimize);}, py::arg("cameras"), py::arg("measurements"), py::arg("rank_tol"), py::arg("optimize"));
[... cut ...] arg("measurements"), py::arg("initialEstimate"));
    m_.def("triangulateNonlinear",[](const gtsam::CameraSetCal3Bundler& cameras, const gtsam::Point2Vector& measurements, const gtsam::Point3& initialEstimate){return gtsam::triangulateNonlinear(cameras, measurements, initialEstimate);}, py::arg("cameras"), py::arg("measurements"), py::arg("initialEstimate"));

// Specializations for STL classes
// TODO(fan): make this automatic
#include "python/gtsam/specializations/geometry.h"

}
```
</details>

ProfFan, 2022-07-13:
Because in the new GTSAM Point3 is really just a typedef, so in Python it becomes a simple numpy array. If you look more closely, you will see in `/python/CMakeLists.txt` these classes are in the ignore list.

ProfFan, 2022-07-13:
They should be shimmed in https://github.com/borglab/gtsam/blob/develop/python/gtsam/__init__.py . 
```
>>> import gtsam
>>> gtsam.Point2()
array([nan, nan])
>>> gtsam.Point2(1, 2)
array([1., 2.])
>>>
```

closed 2022-07-13 by ProfFan

btalb, 2022-07-14:
Thanks @ProfFan; that's very helpful.

I'd already found the ignore list masking out the classes, but realise my import targets weren't correct (hence the shims missing). 

I.e. I had:
```
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.26.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import gtsam

In [2]: gtsam.__file__
Out[2]: '/home/ben/tmp/gtsam/build/python/gtsam/gtsam.cpython-38-x86_64-linux-gnu.so'
```

but it should be:
```
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.26.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import gtsam

In [2]: gtsam.__file__
Out[2]: '/home/ben/tmp/gtsam/build/python/gtsam/__init__.py'
```

ProfFan, 2022-07-14:
Yup, glad that you found out :) I guess this is why PyTorch made their Python binary extension named `C`, lol