borglab/gtsam #1246: MATLAB Wrapper Build fail

ClaudioCimarelli, 2022-07-14:
## Description
I cannot build the Matlab wrapper due to the following errors:
```
/home/claudio/allbuilds/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::pair<gtsam::DiscreteBayesTree, gtsam::DiscreteFactorGraph>::pair(std::pair<boost::shared_ptr<gtsam::DiscreteBayesTree>, boost::shared_ptr<gtsam::DiscreteFactorGraph> >)’
     ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
```
```
/usr/include/c++/8/bits/stl_pair.h:359:38: error: no type named ‘type’ in ‘struct std::enable_if<false, bool>’
                          bool>::type=true>
```
```
/home/claudio/allbuilds/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::pair<gtsam::GaussianConditional, gtsam::JacobianFactor>::pair(std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::JacobianFactor> >)’
```
```
/home/claudio/allbuilds/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::pair<gtsam::GaussianBayesTree, gtsam::GaussianFactorGraph>::pair(std::pair<boost::shared_ptr<gtsam::GaussianBayesTree>, boost::shared_ptr<gtsam::GaussianFactorGraph> >)’
```
Everything else builds correctly, even python wrapper.
## Steps to reproduce

1. `cmake -DCMAKE_BUILD_TYPE=Release  -DGTSAM_BUILD_UNSTABLE:OPTION=ON  -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON -DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX=ON ..`
2. `make`

## Environment

OS: Ubuntu 18.04
GCC  8.4.0
CMAKE 3.23.2
Boost 1.69
Python 3.8.12 (with anaconda)
Matlab 2022a

GTSAM: develop branch (and also latest release 4.1.1)
Thank you for your help.

varunagrawal, 2022-07-21:
This is interesting. I assume GTSAM compiles without issues and you are able to run all the tests?

ClaudioCimarelli, 2022-07-25:
Dear @varunagrawal, the building with Matlab ON stops at the very last percent point.
I have run the `make check` before it once, and it worked as you figured. Of course, once I remove Matlab, everything works normally. I don't know if it may be the GCC version, but it is the only thing I cannot change now.

varunagrawal, 2022-07-26:
Thanks for the info @ClaudioCimarelli. I'll take a look at it today. :)

varunagrawal, 2022-07-28:
@ClaudioCimarelli can you please share the full output of `cmake` and `make`?

ClaudioCimarelli, 2022-07-28:
CMake GUI output:
```
GTSAM Version: 4.2a7
GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
Found Eigen version: 3.3.7
checking for thread-local storage - found
Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
Building 3rdparty
Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
Building base
Building basis
Building geometry
Building inference
Building symbolic
Building discrete
Building hybrid
Building linear
Building nonlinear
Building sam
Building sfm
Building slam
Building navigation
GTSAM Version: 4.2.0
Install prefix: /usr/local
Building GTSAM - shared: ON
Building base_unstable
Building geometry_unstable
Building linear_unstable
Building discrete_unstable
Building dynamics_unstable
Building nonlinear_unstable
Building slam_unstable
Building partition_unstable
GTSAM_UNSTABLE Version: 4.2.0
Install prefix: /usr/local
[1;36mgtwrap[m Package config : /usr/local/lib/cmake/gtwrap
[1;36mgtwrap[m version        : 1.0
[1;36mgtwrap[m CMake path     : /usr/local/lib/cmake/gtwrap
[1;36mgtwrap[m library path   : /usr/local/lib/gtwrap
[1;36mgtwrap[m binary path    : /usr/local/bin/gtwrap
[1;36mgtwrap[m header path    : /usr/local/include/gtwrap
Building wrap module gtsam
Installing Matlab Toolbox to /usr/local/gtsam_toolbox
Building wrap module gtsam_unstable
Installing Matlab Toolbox to /usr/local/gtsam_toolbox
GTSAM_SOURCE_ROOT_DIR: [/home/claudio/git/gtsam]
Wrote /home/claudio/git/gtsam/build/GTSAMConfig.cmake
Wrote /home/claudio/git/gtsam/build/GTSAM_UNSTABLEConfig.cmake
===============================================================
================  Configuration Options  ======================
 CMAKE_CXX_COMPILER_ID type                       : GNU
 CMAKE_CXX_COMPILER_VERSION                       : 8.4.0
 CMake version                                    : 3.23.2
 CMake generator                                  : Unix Makefiles
 CMake build tool                                 : /usr/bin/make
Build flags                                               
 Build Tests                                      : Enabled
 Build examples with 'make all'                   : Enabled
 Build timing scripts with 'make all'             : Disabled
 Build Docs                                       : Enabled
 Build shared GTSAM libraries                     : Enabled
 Put build type in library name                   : Enabled
 Build libgtsam_unstable                          : Enabled
 Build GTSAM unstable Python                      : Disabled
 Build MATLAB Toolbox for unstable                : Enabled
 Build for native architecture                    : Enabled
 Build type                                       : Release
 C compilation flags                              :  -O3 -DNDEBUG
 C++ compilation flags                            :  -O3 -DNDEBUG
 GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_11
 GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_DEBUG               : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_DEBUG           : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_TIMING              : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_TIMING          : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_PROFILING           : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_PROFILING       : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_RELWITHDEBINFO      : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELWITHDEBINFO  : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_MINSIZEREL          : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_MINSIZEREL      : 
 Use System Eigen                                 : OFF (Using version: 3.3.7)
 Use System Metis                                 : OFF
 Use Intel TBB                                    : Yes (Version: 2017.0)
 Eigen will use MKL                               : MKL not found
 Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
 Default allocator                                : TBB
 Cheirality exceptions enabled                    : YES
 Build with ccache                                : No
Packaging flags
 CPack Source Generator                           : TGZ
 CPack Generator                                  : TGZ
GTSAM flags                                               
 Quaternions as default Rot3                      : Disabled
 Runtime consistency checking                     : Disabled
 Rot3 retract is full ExpMap                      : Enabled
 Pose3 retract is full ExpMap                     : Enabled
 Allow features deprecated in GTSAM 4.1           : Enabled
 Metis-based Nested Dissection                    : Enabled
 Use tangent-space preintegration                 : Enabled
MATLAB toolbox flags
 Install MATLAB toolbox                           : Enabled
 MATLAB root                                      : /home/claudio/Programs/MATLAB/R2022a
 MEX binary                                       : /home/claudio/Programs/MATLAB/R2022a/bin/glnxa64/mex
Python toolbox flags                                      
 Build Python module with pybind                  : Disabled
===============================================================
Configuring done
Generating done
```
`make check`  final output:

``` 
100% tests passed, 0 tests failed out of 274

Total Test time (real) =   2.08 sec
[100%] Built target check
```
`make` output file log:
[out.txt](https://github.com/borglab/gtsam/files/9209865/out.txt)

Hi @varunagrawal, I have recompiled everything without python just for this test.

varunagrawal, 2022-08-01:
@ClaudioCimarelli I am still unable to reproduce this. Can you please rebuild with the following cmake command?

```
cmake -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE:OPTION=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON -DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX=ON -DGTSAM_ALLOW_DEPRECATED_SINCE_V42=OFF ..
```

varunagrawal, 2022-08-01:
The only difference between yours and mine is Boost. I use version 1.71.0, so let me try with 1.69 in the meantime.

varunagrawal, 2022-08-01:
I was able to repro this issue and have submitted a fix.

closed 2022-08-01 by varunagrawal

varunagrawal, 2022-08-01:
@ClaudioCimarelli please try the latest develop and if it doesn't work, please reopen this issue. Looking forward to your response.

ClaudioCimarelli, 2022-08-02:
Thank you very much. I suspected boost and installed the custom 1.69,  but reading here around, I thought 1.69 was the best. Should I update it to 1.71?

Thanks again.

varunagrawal, 2022-08-02:
1.71 works extremely well for me, but 1.69 should be fine as well.

PR #1262 (2022-08-01): Fix matlab toolbox compilation
Interface files for Matlab toolbox were missing some namespaces. This PR adds them so that matlab wrapper compilation works as expected.

Fixes #1246