borglab/gtsam #1256: Adding accessible type information to factors

EulersBoiler, 2022-07-28:
This is a conditional feature request, that depends on the actual inability to achieve the functionality i will describe here. I would request the ability for developers to retrieve the name or type of all factors in a graph. Printing a graph using print() and thereby seeing the types is nice, but it does not allow to use this information during execution as far as i see. 

I am working on a visualization Tool to work in the context of ISAM2 and having information about the factor type is crucial to generalize the code. I am currently casting the NonlinearFactors from the graph dynamically into different factors, while try/catch mechanisms cope with segmentation faults when accessing data due to the fact that casting to the wrong type does not throw std::bad_cast exceptions.
Having access to the keys of a factor already allows making guesses on the type of the factor, but an actual explicit source of information would be nice.

The feature could be implemented as easily as adding a name string to the Factor class and initializing the field in the constructor of each factor allows access to the information on the level of the graph.

As initially stated, this feature may be unnecessary as the functionality is already given.

ProfFan, 2022-07-28:
> while try/catch mechanisms cope with segmentation faults when accessing data due to the fact that casting to the wrong type does not throw std::bad_cast exceptions.

I don't think this is the case. Maybe you can post the code you used here and we can see what is the thing you are trying to achieve

EulersBoiler, 2022-08-02:
I experimented some more and realised that dynamic_casts work to determine the type, as the dynamic cast returns a boolean for which one can check. 

My current method to get the constraints from the graph looks like this.

```cpp
bool GraphExtractor::loadGraph(double time, const gtsam::NonlinearFactorGraph& graph) {

    for(const auto& f : graph){

        std::string id = "none";
        if(dynamic_cast<gtsam::PriorFactor<gtsam::Velocity3>*>(f.get()))                         id = "PriorVelocity3";
        else if(dynamic_cast<gtsam::PriorFactor<gtsam::Pose3>*>(f.get()))                        id = "PriorPose3";
        else if(dynamic_cast<gtsam::ImuFactor*>(f.get()) )                                       id = "ImuFactor";
        else if(dynamic_cast<gtsam::BetweenFactor<gtsam::imuBias::ConstantBias>*>(f.get()) )     id = "BetweenFactorBias";
        else if(dynamic_cast<gtsam::GPSFactor*>(f.get()) )                                       id = "GPSFactor";

        std::string name = gtsam::DefaultKeyFormatter(f->keys().at(0));

        if(id == "PriorPose3" && name.rfind('x', 0) == 0){
            addPriorPoseFactor(name, time, f.get());
        }else if(id == "PriorVelocity3" && name.rfind('v', 0) == 0){
            addPriorVelocityFactor(name, time, f.get());
        }else if(id == "ImuFactor"){
            addPreintegratedFactor(name, time, f.get());
        }else if(id == "BetweenFactorBias"){
        }else if(id == "GPSFactor"){
        }else if(id == "none"){
            //printf("Could not load factor.\n");
        }

    }

    return true;
}
```

ProfFan, 2022-08-17:
Yup, you can just dynamic_cast the pointer and check for nullptr.

Feel free to reopen if you think this is not good.

closed 2022-08-17 by ProfFan