borglab/gtsam #1259: Allow using system ceres

Tobias-Fischer, 2022-07-31:
## Feature

Currently gtsam always ships its own bundled version of ceres. It would be great if a similar approach as for Eigen or Metis is taken, where alternatively the system version can be used.

## Motivation

Bundling ceres is undesirable if package managers like conda-forge are used. Also, most distributions ship ceres themselves, so there should not be a need for the bundled version.

## Pitch

Create a cmake/HandleCeres.cmake file that follows the structure of HandleMetis.cmake/HandleEigen.cmake and allows defining GTSAM_USE_SYSTEM_CERES if one wants to use the system ceres.

## Alternatives

N/A

## Additional context

varunagrawal, 2022-08-01:
@Tobias-Fischer I believe this is already tracked in #292? Would you be okay with closing this and updating that issue so we can track all the system libraries together?

Tobias-Fischer, 2022-08-01:
Indeed this issue already mentions Ceres, so happy to close here.

closed 2022-08-01 by Tobias-Fischer