borglab/gtsam #1261: Windows Python bindings do not build (linker errors)

Tobias-Fischer, 2022-07-31:
## Description

## Steps to reproduce

1. Apply this [fix-python-build-win.patch.txt](https://github.com/borglab/gtsam/files/9229802/fix-python-build-win.patch.txt) which is needed to even get to the build error below. Otherwise the build fails much earlier. The patch fixes two issues:

1.1 Basically the problem is that on Linux people rename to the same name different libraries because they have different suffix (like .a vs .so for static vs shared, or .pysomething vs .so for Python vs shared C++). However, on Windows when you build a shared library you create both a name.lib and a name.dll, so if you have a static and shared library with the same name, that results in a collision. Similalry, if you have a name.dll and name.pyd (C++ and Python), both will also generate a name.lib that will result in a collision. See https://github.com/pantor/ruckig/pull/18. (Credits to @traversaro)
1.2 The cmake command `configure_file(${test_file} "${GTSAM_UNSTABLE_MODULE_PATH}/tests/${test_file}" COPYONLY)` leads to a syntax error on Windows for some reason. I am not sure why, for testing I simply commented it out.
2. Build gtsam with `GTSAM_BUILD_PYTHON=ON` on Windows
3. Observe linker errors (just a few listed here; [build.log](https://github.com/borglab/gtsam/files/9229803/build.log)):
```
python\gtsam\gtsam.cp37-win_amd64.pyd : fatal error LNK1120: 105 unresolved externals
...
LINK: command "C:\PROGRA~2\MICROS~1\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe /nologo python\CMakeFiles\gtsam_py.dir\gtsam.cpp.obj python\CMakeFiles\gtsam_py.dir\base.cpp.obj python\CMakeFiles\gtsam_py.dir\geometry.cpp.obj python\CMakeFiles\gtsam_py.dir\linear.cpp.obj python\CMakeFiles\gtsam_py.dir\nonlinear.cpp.obj python\CMakeFiles\gtsam_py.dir\symbolic.cpp.obj python\CMakeFiles\gtsam_py.dir\sam.cpp.obj python\CMakeFiles\gtsam_py.dir\slam.cpp.obj python\CMakeFiles\gtsam_py.dir\sfm.cpp.obj python\CMakeFiles\gtsam_py.dir\navigation.cpp.obj python\CMakeFiles\gtsam_py.dir\basis.cpp.obj /out:python\gtsam\gtsam.cp37-win_amd64.pyd /implib:lib\python_gtsam.lib /pdb:python\gtsam\gtsam.pdb /dll /version:0.0 /machine:x64 /INCREMENTAL:NO -LTCG lib\gtsam.lib C:\hostedtoolcache\windows\Python\3.7.9\x64\libs\python37.lib %PREFIX%\Library\lib\boost_serialization.lib %PREFIX%\Library\lib\boost_system.lib %PREFIX%\Library\lib\boost_filesystem.lib %PREFIX%\Library\lib\boost_thread.lib %PREFIX%\Library\lib\boost_atomic.lib %PREFIX%\Library\lib\boost_date_time.lib %PREFIX%\Library\lib\boost_regex.lib %PREFIX%\Library\lib\boost_timer.lib %PREFIX%\Library\lib\boost_chrono.lib tbb.lib tbbmalloc.lib %PREFIX%\Library\lib\metis.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:python\gtsam\gtsam.cp37-win_amd64.pyd.manifest" failed (exit code 1120) with the following output:
   Creating library lib\python_gtsam.lib and object lib\python_gtsam.exp
geometry.cpp.obj : error LNK2001: unresolved external symbol "public: class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::SO<-1>::vec(class gtsam::OptionalJacobian<-1,-1>)const " (?vec@?$SO@$0?0@gtsam@@QEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@V?$OptionalJacobian@$0?0$0?0@2@@Z)
nonlinear.cpp.obj : error LNK2001: unresolved external symbol "public: virtual class boost::shared_ptr<class gtsam::NonlinearFactor> __cdecl gtsam::LinearContainerFactor::rekey(class std::map<unsigned __int64,unsigned __int64,struct std::less<unsigned __int64>,class std::allocator<struct std::pair<unsigned __int64 const ,unsigned __int64> > > const &)const " (?rekey@LinearContainerFactor@gtsam@@UEBA?AV?$shared_ptr@VNonlinearFactor@gtsam@@@boost@@AEBV?$map@_K_KU?$less@_K@std@@V?$allocator@U?$pair@$$CB_K_K@std@@@2@@std@@@Z)
nonlinear.cpp.obj : error LNK2001: unresolved external symbol "public: virtual class boost::shared_ptr<class gtsam::NonlinearFactor> __cdecl gtsam::LinearContainerFactor::rekey(class std::vector<unsigned __int64,class tbb::detail::d1::tbb_allocator<unsigned __int64> > const &)const " (?rekey@LinearContainerFactor@gtsam@@UEBA?AV?$shared_ptr@VNonlinearFactor@gtsam@@@boost@@AEBV?$vector@_KV?$tbb_allocator@_K@d1@detail@tbb@@@std@@@Z)
nonlinear.cpp.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl gtsam::CustomFactor::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &)const " (?print@CustomFactor@gtsam@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@4@@Z)
nonlinear.cpp.obj : error LNK2001: unresolved external symbol "public: virtual class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::CustomFactor::unwhitenedError(class gtsam::Values const &,class boost::optional<class std::vector<class Eigen::Matrix<double,-1,-1,0,-1,-1>,class std::allocator<class Eigen::Matrix<double,-1,-1,0,-1,-1> > > &>)const " (?unwhitenedError@CustomFactor@gtsam@@UEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@AEBVValues@2@V?$optional@AEAV?$vector@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@V?$allocator@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@std@@@std@@@boost@@@Z)
slam.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::FrobeniusFactor<class gtsam::SO<3> >::FrobeniusFactor<class gtsam::SO<3> >(unsigned __int64,unsigned __int64,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (__imp_??0?$FrobeniusFactor@V?$SO@$02@gtsam@@@gtsam@@QEAA@_K0AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z)
slam.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::FrobeniusFactor<class gtsam::SO<4> >::FrobeniusFactor<class gtsam::SO<4> >(unsigned __int64,unsigned __int64,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (__imp_??0?$FrobeniusFactor@V?$SO@$03@gtsam@@@gtsam@@QEAA@_K0AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z)
slam.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::FrobeniusBetweenFactor<class gtsam::SO<4> >::FrobeniusBetweenFactor<class gtsam::SO<4> >(unsigned __int64,unsigned __int64,class gtsam::SO<4> const &,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (__imp_??0?$FrobeniusBetweenFactor@V?$SO@$03@gtsam@@@gtsam@@QEAA@_K0AEBV?$SO@$03@1@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z)
  Hint on symbols that are defined and could potentially match:
    "__declspec(dllimport) public: __cdecl gtsam::FrobeniusBetweenFactor<class gtsam::SO<3> >::FrobeniusBetweenFactor<class gtsam::SO<3> >(unsigned __int64,unsigned __int64,class gtsam::SO<3> const &,class boost::shared_ptr<class gtsam::noiseModel::Base> const &)" (__imp_??0?$FrobeniusBetweenFactor@V?$SO@$02@gtsam@@@gtsam@@QEAA@_K0AEBV?$SO@$02@1@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@@Z)
slam.cpp.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl gtsam::FrobeniusBetweenFactor<class gtsam::SO<4> >::~FrobeniusBetweenFactor<class gtsam::SO<4> >(void)" (__imp_??1?$FrobeniusBetweenFactor@V?$SO@$03@gtsam@@@gtsam@@UEAA@XZ)
  Hint on symbols that are defined and could potentially match:
    "__declspec(dllimport) public: virtual __cdecl gtsam::FrobeniusBetweenFactor<class gtsam::SO<3> >::~FrobeniusBetweenFactor<class gtsam::SO<3> >(void)" (__imp_??1?$FrobeniusBetweenFactor@V?$SO@$02@gtsam@@@gtsam@@UEAA@XZ)
...
```

## Expected behavior

Build should succeed, and I should be able to `import gtsam` in a Windows python environment without issues.

## Environment

Windows with VS2019 on conda-forge. I expect this to happen on any Windows machine, but I have not confirmed this. Full build commands:
```
mkdir build
cd build

cmake ^
    -GNinja ^
    -DCMAKE_INSTALL_PREFIX=%LIBRARY_PREFIX% ^
    -DGTSAM_BUILD_UNSTABLE:OPTION=ON ^
    -DGTSAM_BUILD_STATIC_LIBRARY=OFF ^
    -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ^
    -DGTSAM_USE_SYSTEM_EIGEN=ON ^
    -DGTSAM_INSTALL_CPPUNITLITE=OFF ^
    -DGTSAM_BUILD_PYTHON=ON ^
    -DGTSAM_USE_SYSTEM_METIS=ON ^
    -DBoost_USE_STATIC_LIBS=OFF ^
    -DBOOST_ROOT="%LIBRARY_PREFIX%" ^
    -DBoost_NO_SYSTEM_PATHS=ON ^
    -DGTSAM_CMAKE_CONFIGURATION_TYPES="Release" ^
    -DCMAKE_BUILD_TYPE="Release" ^
    -DBoost_NO_BOOST_CMAKE=ON ^
    -DPython3_EXECUTABLE=%PYTHON% ^
    -DPython_EXECUTABLE=%PYTHON% ^
    -DPYTHON_EXECUTABLE=%PYTHON% ^
    %SRC_DIR%
```

## Additional information

See https://github.com/conda-forge/gtsam-feedstock/pull/8 where we are able to build gtsam (including gtsam-unstable) successfully, but fail building with python bindings enabled.

varunagrawal, 2022-08-01:
@mikesheffler and I spent a lot of time on Windows compilation, but there is still a lot of work to do. I might have cycles for this after the ICRA deadline...

Tobias-Fischer, 2022-08-01:
Thanks, sounds good!

One thing I forgot to mention is that we only tried with the last release (4.1.1), not with the latest commit on develop where some fixes might have landed.

mikesheffler, 2022-08-02:
I've been solidly on the Linux side since February**, but I can do some test building on Windows as another data point, if that is helpful.

I updated VS 2019 and pulled the `HEAD` of `develop`. Configuring with default options, the `debug` configuration of `gtsam` builds, but `gtsam_unstable` does not. There are a lot of linker errors related to multiply-defined symbols. Just for grins, I thought I would try the `release` configuration, but I forgot that `TangentPreintegration.cpp` takes like an hour (literally) to compile in `release`, so I'm not sure yet -- I would expect to see the same stuff as in `debug`, but it would be good to confirm.

What would actually be helpful for me to do? Should I try to replicate @Tobias-Fischer 's setup or try something else?

-----

** It looks like #1087 is still open. I'm not sure if I still owe anything on that issue? I just kind of wandered off when #1107 landed, and I'm not sure what remains to be done.

Tobias-Fischer, 2023-03-03:
Hiya - I'm just wondering whether you had any time to look into this?

varunagrawal, 2023-03-10:
I'll definitely look into this. @mattking-smith and I were talking about the wrapper on Windows so I have motivation. :)

mattking-smith, 2023-03-13:
> I'll definitely look into this. @mattking-smith and I were talking about the wrapper on Windows so I have motivation. :)

Yeah, I definitely have lots of interest from sponsors using `GTSAM` for Windows wrapping availability. Sponsors have limited Unix OS availability, so @varunagrawal let me know how I can help here

mrw2ee, 2023-04-14:
I'm encountering similar issues both for the python and matlab wrappers. In VisualStudio2019, the project gtsam builds successfully, but not gtsam_py

```
Build started...
1>------ Build started: Project: gtsam_header, Configuration: Release x64 ------
2>------ Build started: Project: pybind_wrap_gtsam, Configuration: Release x64 ------
1>Building Custom Rule C:/gtsam/python/CMakeLists.txt
2>Building Custom Rule C:/gtsam/python/CMakeLists.txt
3>------ Build started: Project: gtsam_py, Configuration: Release x64 ------
3>Building Custom Rule C:/gtsam/python/CMakeLists.txt
3>LINK : fatal error LNK1149: output filename matches input filename 'C:\gtsam\build\lib\Release\gtsam.lib'
3>Done building project "gtsam_py.vcxproj" -- FAILED.
========== Build: 2 succeeded, 1 failed, 3 up-to-date, 0 skipped ==========
```

or gtsam_matlab_wrapper

```
1>   Creating library C:/gtsam/build/wrap/gtsam_mex/Release/gtsam_wrapper.lib and object C:/gtsam/build/wrap/gtsam_mex/Release/gtsam_wrapper.exp
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl gtsam::RedirectCout::str(void)const " (?str@RedirectCout@gtsam@@QEBA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function "void __cdecl gtsamRedirectCout_str_106(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamRedirectCout_str_106@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::RedirectCout::~RedirectCout(void)" (??1RedirectCout@gtsam@@QEAA@XZ) referenced in function "public: __cdecl std::_Temporary_owner<struct gtsam::RedirectCout>::~_Temporary_owner<struct gtsam::RedirectCout>(void)" (??1?$_Temporary_owner@URedirectCout@gtsam@@@std@@QEAA@XZ)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl gtsam::markdown(class gtsam::DiscreteValues const &,class std::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &,class std::map<unsigned __int64,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >,struct std::less<unsigned __int64>,class std::allocator<struct std::pair<unsigned __int64 const ,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > > > const &)" (?markdown@gtsam@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBVDiscreteValues@1@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@3@AEBV?$map@_KV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@U?$less@_K@2@V?$allocator@U?$pair@$$CB_KV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@@std@@@2@@3@@Z) referenced in function "void __cdecl markdown_812(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?markdown_812@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl gtsam::html(class gtsam::DiscreteValues const &,class std::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl(unsigned __int64)> const &,class std::map<unsigned __int64,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >,struct std::less<unsigned __int64>,class std::allocator<struct std::pair<unsigned __int64 const ,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > > > const &)" (?html@gtsam@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBVDiscreteValues@1@AEBV?$function@$$A6A?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_K@Z@3@AEBV?$map@_KV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@U?$less@_K@2@V?$allocator@U?$pair@$$CB_KV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@@std@@@2@@3@@Z) referenced in function "void __cdecl html_815(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?html_815@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: static class std::optional<class gtsam::Pose2> __cdecl gtsam::Pose2::Align(class std::vector<struct std::pair<class Eigen::Matrix<double,2,1,0,2,1>,class Eigen::Matrix<double,2,1,0,2,1> >,class  [... cut ...] truct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamTranslationRecovery_run_4254@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
1>C:\gtsam\build\wrap\gtsam_mex\Release\gtsam_wrapper.mexw64 : fatal error LNK1120: 12 unresolved externals
1>Done building project "gtsam_matlab_wrapper.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 3 up-to-date, 0 skipped ==========
```

Just added PR #1514 which should address 1.2 under original post "Steps to reproduce." Small victory.

ds58, 2023-06-07:
> I've been solidly on the Linux side since February**, but I can do some test building on Windows as another data point, if that is helpful.
>
> I updated VS 2019 and pulled the `HEAD` of `develop`. [...]

Check my comment here if you still experience the long build times on Windows https://github.com/borglab/gtsam/issues/1216#issuecomment-1581163113

Tobias-Fischer, 2023-10-29:
We have now managed to build the Python bindings on Windows. We still need a few patches (see https://github.com/conda-forge/gtsam-feedstock/tree/main/recipe), but I think most of the issues have been addressed in the `dev` branch (we are using the 4.2.0 release in conda-forge).

closed 2023-10-29 by Tobias-Fischer