borglab/gtsam #1264: Attribute error in ISAM2 Python examples (Pose2,Pose3 and Visual)

dars-99, 2022-08-03:
## Description

While running the examples of iSAM2 related to Pose2, Pose3 and Visual, there appears an attribute error in **iSAM2Params()** . It happens at the line
```
parameters.relinearizeSkip = 1
```
Changing the above line to 
```
parameters.setRelinearizeSkip(1)
```
fixes the issue.

varunagrawal, 2022-08-03:
Are you using the pip version or the building the python wrapper from scratch? I just ran `Pose2ISAM2Example.py` and it works well. I thought maybe it was an issue with the .i file but that seems okay too. :confused:

dars-99, 2022-08-03:
I am just cloning the whole repository using git and building the package and directly running it

dars-99, 2022-08-04:
I have installed gtsam using pip3 and the examples are working fine now. But when I run my code  I get an attribute error. 

Currently gtsam is located in the following path **~/.local/lib/python3.6/site-packages/gtsam**.  

I am not sure if this is the right installation path.

dars-99, 2022-08-04:
I think I found out the problem. The installation through pip3 works perfectly because the examples has the change I mentioned above. But the examples on the git doesn't have the change which throws an attribute error.

varunagrawal, 2022-08-04:
I tried running the examples on GitHub directly and they work well for me. Can you please post how you are running your code?

dars-99, 2022-08-04:
Well I tried to remove my installation and rebuilt the package from scratch. The package is built successfully. But I am getting an error saying that the **gtsam** module is not found. Not sure what is the issue. But the install process completed without any error.

dars-99, 2022-08-04:
I tried to rebuild the package from source by completely removing the previous installation. I was able to build the wrappers successfully. But for some reason it is throwing an error saying **gtsam** module not found. I am not sure why this is happening. But the installation completed successfully.

varunagrawal, 2022-08-04:
Did you run `make python-install`?

dars-99, 2022-08-05:
Yep that fixed my issue and also the examples are running without any error. I'm not sure why it is working now. But thanks for your help!

varunagrawal, 2022-08-05:
Ah that's because that command installs the built package so that Python can use it. I am guessing Python was using the pip installed version previously.

Glad we figured this out.

closed 2022-08-05 by varunagrawal