borglab/gtsam #1266: Point3 And Point2 In Python

FaDoSolReLaMiTi, 2022-08-08:
## Description
I managed to compile the Python wrapper but it cannot pass the python-test.
It shows some errors like
> RuntimeError: Attempting to retrieve value with key "10", type stored in Values is class gtsam::GenericValue<class Eigen::Matrix<double,-1,1,0,-1,1> > but requested type was class Eigen::Matrix<double,3,1,0,3,1>

In python, Point3 is 1x3 numpy array and converted into Eigen::Matrix<double,-1,1,0,-1,1> when adding the initial values for the solver. However, the factor graph keeps Eigen::Matrix<double,3,1,0,3,1> when adding a factor (e.g. PriorFactorPoint3).

## Steps to reproduce

1.  Exactly as instructed to install the python wrapper (static link library).

## Expected behavior

It can pass python-test.

## Environment
Window 10
VS2019
Python 3.9

ProfFan, 2022-08-17:
Interesting, I think I fixed this already in #1168. Can you try using `. insertPoint3` and report the results?

FaDoSolReLaMiTi, 2022-08-17:
Thank you very much. It works now. I replaced the `initial_values.insert` with `initial_values.insert_point3` in the python test scripts and then it can pass the python test.

closed 2022-08-17 by FaDoSolReLaMiTi