borglab/gtsam #1269: Tutorial: Error in first order approximation of pose increment?

pglira, 2022-08-12:
[image]

pglira, 2022-08-16:
I noticed right now that it is already corrected in ``doc/ImuFactor.pdf``:
[image]

However, in the turorial on the website it is still not correct.

pglira, 2022-08-16:
I can do that, however, I don't know where to find the tutorial file of the website here on github.

Specifically, I'm talking about this page: https://gtsam.org/tutorials/intro.html

dellaert, 2022-08-16:
This is probably what you’re looking for: https://github.com/borglab/gtsam.org