borglab/gtsam #1272: Custom binary factor python

dars-99, 2022-08-17:
Hello,

I'm trying to create a custom factor between two different kind of variables in python. The variables are camera pose and landmark pose. I am estimating the relative pose between the camera and landmark using a single frame since I know the object's dimension in real world. I am able to get the relative pose but I am unable to add this to the graph because I don't know how to create a custom factor between two different variables in python. The python examples only give an overview for unary factors. It would be nice to know if there is already a function for the above mentioned problem or is there any examples of custom factors for two different kind of variables in python.

Thanks!

ProfFan, 2022-08-17:
I think https://github.com/borglab/gtsam/blob/d3544d4c06c6ebe1af7f4d810c932a7785348d10/python/gtsam/examples/CustomFactorExample.py#L50 is what you need? It takes 2 variables as input.

dars-99, 2022-08-17:
So in this examples the two variables belong to same type(X). I want to pass two different type of variables(X and L). Is it possible?

ProfFan, 2022-08-17:
It's the same.

dars-99, 2022-08-17:
I will look into this. Thanks for your input

closed 2022-08-17 by dars-99