borglab/gtsam #1275: Build failed when using static library 'libgtsam.a'

ChallengeHe, 2022-08-22:
## Description
 Build failed when using static library 'libgtsam.a'. But build with the shared library 'libgtsam.so' has no issues.

## Steps to reproduce
Here is the build output infomation:

`libgtsam.a(LabeledSymbol.cpp.o): In function "boost::io::basic_oaltstringstream<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> >::~basic_oaltstringstream()":
LabeledSymbol.cpp:(.text._ZN5boost2io22basic_oaltstringstreamIcNSt6__ndk111char_traitsIcEENS2_9allocatorIcEEED1Ev[_ZN5boost2io22basic_oaltstringstreamIcNSt6__ndk111char_traitsIcEENS2_9allocatorIcEEED1Ev]+0x30): undefined reference to "__aarch64_ldadd4_acq_rel"
`
 

## Environment
OS: Windows 11

ProfFan, 2022-08-22:
This is related to boost linkage, which boost version are you using? Static build under Windows is something I guess nobody here has ever tried.

ChallengeHe, 2022-08-22:
Thanks a lot for your reply.My version is "boost-1_74". Actually I am tring to use this lib on Android platform and it was build by NDK . The static lib was built on Ubuntu linux OS. I guess it could be platform or compiler issue. -------- 原始邮件 --------发件人： Fan Jiang ***@***.***>日期： 2022年8月23日周二 01:02收件人： borglab/gtsam ***@***.***>抄送： Damon ***@***.***>, Author ***@***.***>主    题： Re: [borglab/gtsam] Build failed when using static library 'libgtsam.a' (Issue #1275)
This is related to boost linkage, which boost version are you using? Static build under Windows is something I guess nobody here has ever tried.

ProfFan, 2022-08-22:
Is dynamic linkage a problem for you? Last time I tried static linking on Linux there is no problem. `__aarch64_ldadd4_acq_rel` looks like an intrinsic so I guess you can try turn off `-march=native` and try again.

Please check your compilers.

ChallengeHe, 2022-08-23:
> 

Dynamic lib works fine , but I need to use static lib for my Project. 

Here is my commands for compiling:
`/opt/clion-2022.1.3/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/opt/android/sdk/ndk/25.0.8775105/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android30-clang -DCMAKE_CXX_COMPILER=/opt/android/sdk/ndk/25.0.8775105/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android30-clang++ -DCMAKE_TOOLCHAIN_FILE=/opt/android/sdk/ndk/25.0.8775105/build/cmake/android.toolchain.cmake -DANDROID_NDK=/opt/android/sdk/ndk/25.0.8775105 -DANDROID_ABI=arm64-v8a -DANDROID_PLATFORM=android-30 -DANDROID_STL=c++_shared    -G "CodeBlocks - Unix Makefiles" -DEIGEN3_INCLUDE_DIR=/usr/include/eigen3`

Finnally I dont know how to turn off `-march=native` option, and no search results for it in the library codes.

Thank you again.

ProfFan, 2022-08-23:
`-DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF` in cmake

9SL9, 2023-03-14:
I am also trying to build a static library on macOS. I'm using the following command but it isn't working and shows the following messages below. Any ideas?
--
Build shared GTSAM libraries                     : Enabled
--
CMake Warning:
  Manually-specified variables were not used by the project:

    GTSAM_BUILD_STATIC_LIBRARY
    GTSAM_INSTALL_STATIC_LIBS
--

cmake .. -DGTSAM_BUILD_STATIC_LIBRARY=ON -DGTSAM_BUILD_SHARED_LIBS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_UNSTABLE=OFF -DGTSAM_USE_SYSTEM_EIGEN=ON -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/xx/Downloads/gtsam-static/build/ss

ProfFan, 2023-03-14:
Maybe you are not using the correct arguments?

dellaert, 2023-06-04:
Closing as most likely not systematic.

closed 2023-06-04 by dellaert