borglab/gtsam #1278: Bug when serializing binary ISAM2 with large factor graph.

wangg12, 2022-08-23:


ProfFan, 2022-08-23:
Please provide information as suggested by the template

wangg12, 2022-08-23:
Just add a large number of factors using the test example provided in this repo, you will see a core dump.

ProfFan, 2022-08-23:
Sorry but if you do not provide stack traces and your system info/build info we cannot help.

wangg12, 2022-08-23:
@ProfFan I added an example repo https://github.com/wangg12/gtsam_serialze_test to reproduce the bug.

On different computers, the `num` limit https://github.com/wangg12/gtsam_serialze_test/blob/master/test.cc#L77 for `Segmentation fault (core dumped)` might be different.

I used gdb to debug this bug, but it is hard to know how to fix this problem:
```
Thread 1 "test_gtsam_seri" received signal SIGSEGV, Segmentation fault.
0x00007ffff7977e16 in ?? () from /lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0
(gdb) bt
#0  0x00007ffff7977e16 in ?? () from /lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0
#1  0x00007ffff7963606 in boost::archive::detail::basic_oarchive::save_object(void const*, boost::archive::detail::basic_oserializer const&) ()
   from /lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0
#2  0x00005555556b3145 in boost::archive::detail::save_non_pointer_type<boost::archive::binary_oarchive>::save_standard::invoke<gtsam::JacobianFactor> (ar=..., t=...)
    at /usr/include/boost/archive/detail/oserializer.hpp:258
#3  0x00005555556ac83c in boost::archive::detail::save_non_pointer_type<boost::archive::binary_oarchive>::invoke<gtsam::JacobianFactor> (ar=..., t=...)
    at /usr/include/boost/archive/detail/oserializer.hpp:315
#4  0x00005555556a5a90 in boost::archive::save<boost::archive::binary_oarchive, gtsam::JacobianFactor const> (ar=..., t=...) at /usr/include/boost/archive/detail/oserializer.hpp:539
#5  0x000055555569ef9e in boost::archive::detail::common_oarchive<boost::archive::binary_oarchive>::save_override<gtsam::JacobianFactor const> (this=0x7fffffffbf10, t=...)
    at /usr/include/boost/archive/detail/common_oarchive.hpp:71
#6  0x0000555555698ac1 in boost::archive::basic_binary_oarchive<boost::archive::binary_oarchive>::save_override<gtsam::JacobianFactor> (this=0x7fffffffbf10, t=...)
    at /usr/include/boost/archive/basic_binary_oarchive.hpp:80
#7  0x00005555556935f9 in boost::archive::binary_oarchive_impl<boost::archive::binary_oarchive, char, std::char_traits<char> >::save_override<gtsam::JacobianFactor const> (
    this=0x7fffffffbf10, t=...) at /usr/include/boost/archive/binary_oarchive_impl.hpp:59
#8  0x000055555568e498 in boost::archive::detail::interface_oarchive<boost::archive::binary_oarchive>::operator<< <gtsam::JacobianFactor> (this=0x7fffffffbf10, t=...)
    at /usr/include/boost/archive/detail/interface_oarchive.hpp:70
#9  0x0000555555688e3f in boost::serialization::save<boost::archive::binary_oarchive, gtsam::JacobianFactor> (ar=..., t=...) at /usr/include/boost/serialization/nvp.hpp:48
#10 0x00005555556837fa in boost::serialization::free_saver<boost::archive::binary_oarchive, boost::serialization::nvp<gtsam::JacobianFactor> >::invoke (ar=..., t=..., file_version=0)
    at /usr/include/boost/serialization/split_free.hpp:45
#11 0x000055555567df5c in boost::serialization::split_free<boost::archive::binary_oarchive, boost::serialization::nvp<gtsam::JacobianFactor> > (ar=..., t=..., file_version=0)
    at /usr/include/boost/serialization/split_free.hpp:74
#12 0x0000555555678811 in boost::serialization::serialize<boost::archive::binary_oarchive, gtsam::JacobianFactor> (ar=..., t=..., file_version=0)
    at /usr/include/boost/serialization/nvp.hpp:65
#13 0x00005555556716a4 in boost::serialization::serialize_adl<boost::archive::binary_oarchive, boost::serialization::nvp<gtsam::JacobianFactor> > (ar=..., t=..., file_version=0)
    at /usr/include/boost/serialization/serialization.hpp:118
#14 0x000055555566664e in boost::archive::detail::save_non_pointer_type<boost::archive::binary_oarchive>::save_only::invoke<boost::serialization::nvp<gtsam::JacobianFactor> > (ar=..., 
    t=...) at /usr/include/boost/archive/detail/oserializer.hpp:246
#15 0x0000555555657a3c in boost::archive::detail::save_non_pointer_type<boost::archive::binary_oarchive>::invoke<boost::serialization::nvp<gtsam::JacobianFactor> > (ar=..., t=...)
    at /usr/include/boost/archive/detail/oserializer.hpp:315
#16 0x0000555555645a57 in boost::archive::save<boost::archive::binary_oarchive, boost::serialization::nvp<gtsam::JacobianFactor> const> (ar=..., t=...)
    at /usr/include/boost/archive/detail/oserializer.hpp:539
#17 0x0000555555635cde in boost::archive::detail::common_oarchive<boost::archive::binary_oarchive>::save_override<boost::serialization::nvp<gtsam::JacobianFactor> const> (
    this=0x7fffffffbf10, t=...) at /usr/include/boost/archive/detail/common_oarchive.hpp:71
#18 0x0000555555628f41 in boost::archive::basic_binary_oarchive<boost::archive::binary_oarchive>::save_override<boost::serialization::nvp<gtsam::JacobianFactor> > (this=0x7fffffffbf10, 
    t=...) at /usr/include/boost/archive/basic_binary_oarchive.hpp:80
#19 0x0000555555619b69 in boost::archive::binary_oarchive_impl<boost::archive::binary_oarchive, char, std::char_traits<char> >::save_override<boost::serialization::nvp<gtsam::JacobianFactor> const> (this=0x7fffffffbf10, t=...) at /usr/include/boost/archive/binary_oarchive_impl.hpp:59                                                                                                
#20 0x0000555555608c00 in boost::archive::detail::interface_oarchive<boost::archive::binary_oarchive>::operator<< <boost::serialization::nvp<gtsam::JacobianFactor> > (this=0x7fffffffbf10, 
    t=...) at /usr/include/boost/archive/detail/interface_oarchive.hpp:70
[... cut ...]
#24 0x0000555555658d91 in boost::serialization::serialize<boost::archive::binary_oarchive, gtsam::GaussianConditional> (ar=..., t=..., file_version=0)
    at /usr/include/boost/serialization/serialization.hpp:68
#25 0x0000555555647171 in boost::serialization::serialize_adl<boost::archive::binary_oarchive, gtsam::GaussianConditional> (ar=..., t=..., file_version=0)
--Type <RET> for more, q to quit, c to continue without paging--q
```

**NOTE**: For the same `num`, I can save the factor graph and values to `.xml` without any problem.

ProfFan, 2022-08-23:
Can you try a different boost version? Also, what is your stack size limit? It seems this is a stack overflow. (you can use ulimit to adjust)

wangg12, 2022-08-23:
tried, doesnt work.

iwohaocai, 2022-09-19:
@wangg12 i have the same problem ,have you solved?

wangg12, 2022-09-19:
Of course not.

iwohaocai, 2022-09-19:
@wangg12  我在自己查，看看能不能自己搞定，要不要加个 v

wangg12, 2022-09-19:
@iwohaocai ok, my id is `posegraph`

ProfFan, 2022-09-19:
Might be a boost version issue

iwohaocai, 2022-09-21:
@ProfFan @wangg12 thanks for your help ,two reasons caused this error,first if we have too many factor graph ,we should 
ulimit -s unlimited . the second reason is i use two noise ,   noiseModel::Robust and noiseModel::Diagonal,and i did not BOOST_CLASS_EXPORT_GUID(gtsam::noiseModel::mEstimator::Cauchy, "gtsam_noiseModel_mEstimator_Cauchy").
my english is poor ,thank you again

ProfFan, 2022-09-21:
Glad to see it resolved :)

closed 2022-09-21 by ProfFan