borglab/gtsam #1279: writeG2o will eliminate symbols

hiyyg, 2022-08-24:
If more than one symbols are in the factor graph, the vertex indices in the results will only contain the keys without the symbols. 

So there will be duplicated indices which is not be expected behavior.

xqyuan060728, 2022-09-27:
I have come across the same problem! 
Besides, I have tested the code with different version of gtsam, I found that in gtsam4.0.2(the version lvi-sam recommented), the symbols and keys can be stored as long sequences of figures in a g2o file, however, in gtsam4.2.0, the symbol is missing, there are only keys of vertex in the g2o file. So, when load the g2o file, it throwing an instance of 'gtsam::ValuesKeyAlreadyExists'. 
There is a note to this function(writeG2o):
'Note:behavior change in PR #471: to be consistent with load2D and load3D, we write the indices to file and not the full Keys. This change really only affects landmarks, which get read as indices but stored in values with the symbol L(index).'
(https://gtsam.org/doxygen/a01596.html#a68bc584d72da3747666ac134a9490f92).
But I cannot find the effective way to solve this question in gtsam4.2.0.

varunagrawal, 2023-06-16:
@hiyyg @xqyuan060728 can you please provide with an example that fails? We can use that to reproduce and understand the issue better.

varunagrawal, 2023-09-04:
Closing due to lack of a response. Please reopen if the problem still persists.

closed 2023-09-04 by varunagrawal