borglab/gtsam #1283: wrong use of gtsam_wrapper, undefined function 'gtsam.KeyVector' corresponding to input argument of type 'uint64' when running test_gtsam.m

duanabao, 2022-08-29:
The environment is:
vmware + ubuntu 18.04 + gtsam of the latest develop branch + matlab 2021b

The problem description: matlab toolbox is installed succefully and I get the gtsam_toolbox in /usr/local, but when I tried to run /usr/local/gtsam_toolbox/gtsam_tests/test _gtsam

I got the following messages:
*****************************************
wrong use of gtsam_wrapper
Exception from gtsam:
undefined function corresponding to input argument of type 'uint64'
'gtsam.KeyVector'.
Wrong use of gtsam_wrapper
Exception from gtsam:
Undefined function 'gtsam.KeyVector' corresponding to input arguments of type 'uint64'.

Error gtsam.utilities.createKeyVector (line 3)
        varargout{1} = gtsam_wrapper(67, varargin{:});

Error testUtilities (line 28)
actual = utilities.createKeyVector([1;2;3]);

Error test_gtsam (line 49)
testUtilities

******************************************

I tried to install gtsam downloaded the master branch, but the installation failed. Thanks very much

varunagrawal, 2022-12-24:
We're planning a release soon. Hopefully this issue is fixed in that release (I made a lot of fixes to the matlab wrapper).

varunagrawal, 2023-03-10:
@duanabao can you please try with the latest version and report on whether it works?

closed 2023-04-27 by varunagrawal