borglab/gtsam #1297: gtsam should use eigen cmake config file

OznOg, 2022-09-27:
## Feature

gtsam uses a custom FindEigen3.cmake but eigen3 already provides a Eigen3Config.cmake for eigen detection.

## Motivation

Using vanilla cmake detection will prevent maintaining code and overall prevent bad usage and detection of the library:

for example, path to the eigen lib eventually get embeded in the gtsamConfig.cmake which should not be the case (gtsam should use find_dependency() macro instead so it would not break when using packaged version of eigen)

Moreover, gtsam shoud use Eigen3::eigen library names instead of forcing eigen include path.

## Alternatives

have an hybrid approach trying to use the vanilla eigenConfig of the system and fallback on the custom one if nothing is found

## Additional context

Useful when working with system eigen and trying to package gtsam (rpm, deb, conan etc...)

ProfFan, 2022-10-04:
Yeah but we don't have bandwidth for this, but contributions are always welcome!

OznOg, 2022-10-05:
what is the oldest eigen version supposed to be supported by current gtsam?

OznOg, 2022-10-06:
looks like to be 3.3.0, any one can confirm, please?

ProfFan, 2022-10-06:
3.3.4-4 as in Ubuntu 18.04

varunagrawal, 2022-10-07:
@OznOg has this been a problem for you? This seems to be more of an opinion rather than a bug or issue.

OznOg, 2022-10-08:
This is not, I'm working to integrate gtsam on yocto and I had to patch it for this, this is why I'm looking forward having the code patched upstream.
Moreover *any* packager will encounter the very same problem as the path put in the includes of the target references the build tree which it should not.

varunagrawal, 2022-10-08:
Awesome. Thanks for the info and the PR @OznOg :)

OznOg, 2022-10-09:
welcome! If I write more patches for yocto tht make sens in the mainline, I'll open another ticket.

closed 2022-10-25 by varunagrawal