borglab/gtsam #1303: About [mapOptmization-6] process has died

Neal0529, 2022-10-07:
Because I'm the new computer with the version ubuntu 18.04.I have checked my version(gtsam 4.0.0-alpha2 ). I never meet the problem if I make "sudo apt-get install libparmetis-dev" or "export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib".
The more detalis like this:
SUMMARY
========

PARAMETERS
 * /mapOptmization/file_dir: /home/***
 * /mapOptmization/pitch: 0.0
 * /mapOptmization/roll: 1.570795
 * /mapOptmization/x: 0.0
 * /mapOptmization/y: 0.0
 * /mapOptmization/yaw: 1.570795
 * /mapOptmization/z: 0.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /use_sim_time: False

NODES
  /
    base_link_to_camera (tf/static_transform_publisher)
    camera_init_to_map (tf/static_transform_publisher)
    featureAssociation (lego_loam/featureAssociation)
    imageProjection (lego_loam/imageProjection)
    mapOptmization (lego_loam/mapOptmization)
    rviz (rviz/rviz)
    transformFusion (lego_loam/transformFusion)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [357]
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.
process[camera_init_to_map-2]: started with pid [358]
process[base_link_to_camera-3]: started with pid [360]
process[imageProjection-4]: started with pid [366]
[ INFO] [1665144927.330510947]: ----> Image Projection Started.
process[featureAssociation-5]: started with pid [372]
[ INFO] [1665144927.394488015]: ----> Feature Association Started.
process[mapOptmization-6]: started with pid [382]
double free or corruption (out)
process[transformFusion-7]: started with pid [391]
[ INFO] [1665144927.522901869]: ----> Transform Fusion Started.
[mapOptmization-6] process has died [pid 382, exit code -6, cmd /home/000/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/a/.ros/log/5a7a1b74-4639-11ed-9622-d45d6436bf20/mapOptmization-6.log].
log file: /home/a/.ros/log/5a7a1b74-4639-11ed-9622-d45d6436bf20/mapOptmization-6*.log

maybe the problem is "process[mapOptmization-6]: started with pid [382] double free or corruption (out)"?
I relly need you help!

ProfFan, 2022-10-07:
Please report this problem to LEGO LOAM

closed 2022-10-07 by ProfFan

1uxxx, 2022-10-25:
I encountered the same problem and wanted to know how to solve it.

The difference is that my lego loam did not run with this error at the beginning.

But when I run lego loam after installing the dependencies of hdl_localization, it will report this error

Neal0529, 2022-10-25:
I just reinstall ubuntu 20.04
 

琉年欢
***@***.***