borglab/gtsam #1307: Sliding window with ISAM2

ArisKoublis, 2022-10-13:
Trying to implement a Graph SLAM with a sliding window optimization scheme, meaning that I want the poses and the landmarks of the last n seconds to be optimized only during each update, not the entirety of the factors but also not wiping the factor graph clean after each update (updates happen more regularly than n seconds). Can anybody please propose a computationally efficient way of removing old poses and their factors using ISAM2? 
For the moment being I have tried doing that with the removeFactorIndices argument but it always leads to the same error:
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)

The part of the code leading to that error is:
gtsam::ISAM2UpdateParams params2;
gtsam::FactorIndices f;
f.push_back(0);
f.push_back(1);
params2.removeFactorIndices = f;
isam2_ = gtsam::ISAM2();
isam2_.update(graph, est_state_, params2);

A simple update (without params2) is working just fine.

kechaofan, 2022-11-11:
take a look at incremental_smoother

varunagrawal, 2022-11-12:
What you want is a fixed lag smoother. You want to marginalize out the last pose/landmarks when you add new ones.

varunagrawal, 2022-12-24:
Closing as resolved. Please ask general questions on the Google Group for faster responses.

closed 2022-12-24 by varunagrawal