borglab/gtsam #1325: fisheye camera distortion issue

huanglilong, 2022-11-13:
https://github.com/borglab/gtsam/blob/256c664aaaa27a3204411e461d4d93c0e168bd6b/gtsam/geometry/Cal3Fisheye.cpp#L57

@ghaggin according to https://arxiv.org/pdf/1807.08957.pdf and https://users.aalto.fi/~kannalj1/publications/tpami2006.pdf

`const double s = scaling * K.dot(T);` should be `const double s = scaling * K.dot(T) * t;`

because K.dot(T)  is not equals to d(t) in (27)
[image]

cc: @dellaert @varunagrawal

huanglilong, 2022-11-13:
the issue is from this pull https://github.com/borglab/gtsam/pull/307 @ghaggin @dellaert 
[image]

varunagrawal, 2022-11-13:
That's interesting. Looks like we don't have the correct unit test coverage for this.

huanglilong, 2022-11-13:
from kalibr: https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/EquidistantDistortion.hpp#L23
[image]

huanglilong, 2022-11-14:
@varunagrawal @ghaggin @roderick-koehle @dellaert please review https://github.com/borglab/gtsam/pull/1326

closed 2022-11-16 by huanglilong