borglab/gtsam #1329: TBB library not used when building in RelWithDebInfo

berndpfrommer, 2022-11-16:
This bug/feature  is observed when building on Ubuntu 20.04LTS x86_64 as "RelWithDebInfo". Since RelWithDebInfo is the current mode how the Ubuntu packages are built, this bug/feature causes the Ubuntu package to compile without TBB.
## Description
When running 
```
cmake ..
```
The TBB library is found and used as it should. In the cmake output the following lines can be found:
```
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
```
When building with RelWithDebInfo:
```
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebINfo
```
The TBB library is ignored, presumably because it's not the debug version:
```
-- Could NOT find TBB (missing: tbb tbbmalloc) (found suitable version "2020.1", minimum required is "4.4")  
```
I found a similar issue here but the discussion digresses into Cmake internals that I don't understand:

https://github.com/AcademySoftwareFoundation/openvdb/issues/1269

ProfFan, 2023-01-11:
I think this is related to TBB providing two versions of their libraries (Debug and Release) and the `FindTBB.cmake` does not know how to handle RelWithDebInfo

varunagrawal, 2023-06-09:
@berndpfrommer I updated TBB in a recent PR. Can you please try with the latest develop and see if this works for you?
I am able to get TBB linked with RelWithDebInfo on a Macbook.

berndpfrommer, 2023-06-10:
Cmake output (when run as ``cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo``) with latest dev branch (commit adc438922017d7ca986e03d7d6db35cb0134817b) on Ubuntu 20.04:
```
Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc
```
and on Ubuntu 22.04:
```
Found TBB: /usr/include (found suitable version "2021.5", minimum required is "4.4") found components: tbb tbbmalloc
```
Consider this fixed as of the commit mentioned above.
Thanks!!!

closed 2023-06-10 by berndpfrommer