borglab/gtsam #1333: Pose3 Subvertices

hovavalon, 2022-11-23:
## Feature

Pose3 contains both 3D transformation and 3D attitude. It should be possible to add a factor which is attached only to a pose's transformation or attitude.

## Motivation

In some cases, we have a Pose3 element, and we would like to add constraints only on its attitude or on its translation

## Pitch

I am not that familiar with the code of the library, so I will present a few general ideas about how I think this can be done, and would happily be corrected.

* Currently the `rotation` and `translation` functions of `Pose3` return a constant reference. Creating a function which returns a non-constant reference will enable passing the attributes to other functions to be changed by them. This will work only if while creating new keys, the reference is saved and not a copy of it.
* New classes, `Pose3Rot : public Rot3` and `Pose3Point : Point3` may be created, which obtain a reference to a `Pose3` object and return as their data the values from `Pose3::rotation`, `Pose3::translation` respectively.

If there are better ways to approach this problem I would be happy to know, and given general guidelines I will work on a pull request adding this feature myself.

dellaert, 2022-11-27:
There is already an Attitude factor, and other elements can be constrained using expression factors.

hovavalon, 2022-11-27:
OK, great!
I am not sure I fully understand, how would I use expressions in such a case?
Should I create a key for the pose, the translation and the attitude, and then generate equality expressions between them somehow?

varunagrawal, 2023-03-10:
With expressions, you can define your operations such as
```cpp
Pose3_ p_;

graph.addExpressionFactor(p_.translation, measured_translation, noise_model);
```

Here, I've added an expression to assign a prior to only the translation part of the Pose3, using a measured translation value `measured_translation`.

Check out `SFMExampleExpressions.cpp` for a more detailed example.

varunagrawal, 2023-03-10:
Please feel free to continue the discussion on the Google Group.

closed 2023-03-10 by varunagrawal