borglab/gtsam #1334: IMUScaledFactor

hovavalon, 2022-11-23:
## Feature

A factor named `IMUScaledFactor`, which gets a key for a scale , two keys for two positions, and the IMU preintegration between them. 

## Motivation

Some SLAM algorithms such as ORB-SLAM3 use bundle adjustment to find out the scale of the 3D positions of points identified in multiple frames.
Without this scale, it is not practical to obtain accelerometer biases from the bundle adjustment.

## Pitch

I propose a class named `IMUScaledFactor`, with a constructor `IMUScaledFactor(Key pose_i, Key vel_i, Key pose_j, Key vel_j, Key bias, Key scale, const PreintegratedImuMeasurements& pim)`, in which the error is computed as in `IMUFactor`, but with the translation of the  poses multiplied by the value of the scale.
The Jacobians will be updated accordingly.

I am beginning to work on an implementation myself, and would be happy to hear recommendation/suggestions/warnings which you may think are relevant.

dellaert, 2022-11-27:
Hmmm. There are observability issues with IMUs, for sure, but under the right motions biases should be observable without introducing an unknown scale, is my understanding.

hovavalon, 2022-11-27:
For sure, first approximating the bias and then rescaling the visual data accordingly is possible, but it has some disadvantages.
Actually, I would like to use something like the scale factor used by the ORB-SLAM3 [here](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/0df83dde1c85c7ab91a0d47de7a29685d046f637/src/G2oTypes.cc), `EdgeInertialGS`, as ORB-SLAM3 have shown substantial improvement over other methods using separated bundle adjustment phases for visual and inertial data.

dellaert, 2022-12-01:
Sorry, very little bandwidth :-( 
I'll read the ORB-SLAM 3 paper this week and share my thoughts

laxnpander, 2023-03-02:
I'd like to pick this discussion up. As far as I understand the orb slam 3 paper they never estimate poses, velocities AND scale in one optimisation. The picture explains it quite nicely:

[image]

For me personally scenario c) is the most interesting at the moment. In this estimation they keep the visual poses with arbitrary scale fixed, but estimate scale, velocities, biases and gravitational vector. This is more or less an initialisation optimisation to align the visual with the imu/body frame. 

I am still pretty much lost in the GTSAM code, but I will try to implement a solution for this. In my head this sounds like a  modified version of the IMUFactor.

laxnpander, 2023-03-20:
I now implemented a version which is almost a 1:1 copy. However, the error of the LM solver goes straight up to infinity. I am not yet sure why, but will investigate. I guess some Jacobian is wrong? 

```c
Vector9 ImuScaleFactor::computeErrorAndJacobians(
    const Pose3& pose_i,
    const Vector3& v_i,
    const Pose3& pose_j,
    const Vector3& v_j,
    const imuBias::ConstantBias& b_i,
    const double& s,
    const Rot3& Rwg,
    OptionalJacobian<9, 9> H1,
    OptionalJacobian<9, 3> H2,
    OptionalJacobian<9, 9> H3,
    OptionalJacobian<9, 3> H4,
    OptionalJacobian<9, 6> H5,
    OptionalJacobian<9, 1> H6,
    OptionalJacobian<9, 2> H7) const
{
  double dt = _PIM_.deltaTij();

  PreintegrationParams& params = _PIM_.p();

  // -----------------------------------------------------------------------------
  // #2 Compute error

  // Extract uncorrected deltas
  Rot3 dR = _PIM_.deltaRij();
  Vector3 dP = _PIM_.deltaPij();
  Vector3 dV = _PIM_.deltaVij();

  const Vector3& Pwb_i = pose_i.translation();
  const Rot3& Rwb_i = pose_i.rotation();
  const Rot3  Rbw_i = Rot3(Rwb_i.transpose());

  const Vector3& Pwb_j = pose_j.translation();
  const Rot3& Rwb_j = pose_j.rotation();

  Vector3 eR = Rot3::Logmap(Rot3(dR.transpose()*Rbw_i.matrix()*Rwb_j.matrix()));
  Vector3 eV = Rbw_i*(s*(v_j-v_i) - Rwg*params.n_gravity*dt) - dV;
  Vector3 eP = Rbw_i*(s*(Pwb_j-Pwb_i-v_i*dt) - 0.5*Rwg.matrix()*params.n_gravity * dt * dt) - dP;

  Vector9 error;
  error << eR, eV, eP;

  // -----------------------------------------------------------------------------
  // #3 Compute Jacobians

  Matrix32 G;
  G << 0.0, -9.81,
      9.81, 0.0,
      0.0, 0.0;

  imuBias::ConstantBias db = _PIM_.biasHat()-b_i;
  const Vector3 &dbg = db.gyroscope();

  Matrix3 invJr = SO3::ExpmapDerivative(eR);
  Matrix3 JRg = _PIM_.delRdelBiasOmega();
  Matrix3 JVg = _PIM_.delVdelBiasOmega();
  Matrix3 JPg = _PIM_.delPdelBiasOmega();
  Matrix3 JVa = _PIM_.delVdelBiasAcc();
  Matrix3 JPa = _PIM_.delPdelBiasAcc();

  if(H1)
  {
    H1->setZero();
    H1->block<3,3>(0, 0) = -invJr*Rwb_j.transpose()*Rwb_i.matrix();
    H1->block<3,3>(3, 0) = spear::SO3d::hat(Rbw_i*(s*(v_j-v_i) - params.n_gravity*dt));
    H1->block<3,3>(6, 0) = spear::SO3d::hat(Rbw_i*(s*(Pwb_j-Pwb_i-v_i*dt) - 0.5*params.n_gravity*dt*dt));
    H1->block<3,3>(6, 3) = Eigen::DiagonalMatrix<double,3>(-s,-s,-s);
    //std::cout << "H1:\n" << *H1 << std::endl;
  }

  if(H2)
  {
    H2->setZero();
    H2->block<3,3>(3, 0) = -s*Rbw_i.matrix();
    H2->block<3,3>(6, 0) = -s*Rbw_i.matrix()*dt;
    //std::cout << "H2:\n" << *H2 << std::endl;
  }

  if(H3)
  {
    H3->setZero();
    H3->block<3,3>(0, 0) = invJr;
    H3->block<3,3>(6, 3) = s*Rbw_i.matrix()*Rwb_j.matrix();
    //std::cout << "H3:\n" << *H3 << std::endl;
  }

  if(H4)
  {
    H4->setZero();
    H4->block<3,3>(3, 0) = s*Rbw_i.matrix();
    //std::cout << "H4:\n" << *H4 << std::endl;
  }

  if(H5)
  {
    H5->setZero();
    H5->block<3,3>(3, 0) = -JVa;
    H5->block<3,3>(6, 0) = -JPa;
    H5->block<3,3>(0, 3) = -invJr*(dR.transpose()*Rbw_i.matrix()*Rwb_j.matrix()).transpose()*SO3::LogmapDerivative(JRg*dbg)*JRg;
    H5->block<3,3>(3, 3) = -JVg;
    H5->block<3,3>(6, 3) = -JPg;
    //std::cout << "H5:\n" << *H5 << std::endl;
  }

  if (H6)
  {
    H6->setZero();
    H6->block<3,1>(3,0) = Rbw_i*(v_j-v_i);
    H6->block<3,1>(6,0) = Rbw_i*(Pwb_j-Pwb_i-v_i*dt);
    //std::cout << "H6:\n" << *H6 << std::endl;
  }

  if (H7)
  {
    H7->setZero();
    H7->block<3,2>(3,0) = -Rbw_i.matrix()*Rwg.matrix()*G*dt;
    H7->block<3,2>(6,0) = -0.5*Rbw_i.matrix()*Rwg.matrix()*G*dt*dt;
    //std::cout << "H7:\n" << *H7 << std::endl;
  }

  //std::cout << "s: " << s << std::endl;
  //std::cout << "e: " << error.transpose() << std::endl;
  //std::cout << "P: " << pose_i << std::endl;
  //std::cout << "V: " << v_i << std::endl;

  return error;
}
```

dellaert, 2023-06-04:
This is a great thread. @ankur-rc we might want to revisit the discussion here at some point. Closing for now as no plans imminent on providing this functionality.

closed 2023-06-04 by dellaert