borglab/gtsam #1342: Python warpped install error:pyparsing.exceptions.ParseException

Wjt-shift, 2022-12-15:
I want to install gtsam in anaconda virtual enviroment.use follow commend:
**mkdir <gtsam_dir>/build
cd <gtsam_dir>/build
cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.9.15 -DCMAKE_INSTALL_PREFIX=/home/wjt/software/anaconda3/envs/slam
make python-install**
And get this error.I don't know what cause it.I try install diferent version pyparsing.But can't solve it.Can someobe help me?Thank you a lot!
[image]

varunagrawal, 2022-12-24:
What version of pyparsing do you have installed?

hjxwhy, 2022-12-27:
@varunagrawal hello, I have also the same problem, I have tried pyparsing 2.4.2 and 3.0.9, the problem can not be solved.

ProfFan, 2022-12-27:
@hjxwhy Could you provide the output of you entire shell history?

Also please provide your CMakeCache.txt contents in a gist.

Thank you!

VladDok, 2022-12-28:
May be we have similar issues.
[image]

varunagrawal, 2022-12-29:
@VladDok @hjxwhy we will need more information than that please.
Please give us your OS, your python version, the version of pyparsing and any other relevant information (e.g. did you change any files?).

VladDok, 2022-12-29:
@varunagrawal, additional information:
**Ubuntu 22.04.01
Python 3.9.12
pyparsing 3.0.9

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Tue_Mar__8_18:18:20_PST_2022
Cuda compilation tools, release 11.6, V11.6.124
Build cuda_11.6.r11.6/compiler.31057947_0**

No, I didn't change anything.

varunagrawal, 2022-12-29:
And are you using the develop branch of GTSAM?

VladDok, 2022-12-29:
I am using MAIN.

varunagrawal, 2022-12-29:
@VladDok @Wjt-shift it seems both of you are trying to compile NeRF-SLAM. This seems to be an issue with that codebase and not GTSAM.
Have you tried cloning GTSAM separately and compiling it?

varunagrawal, 2022-12-29:
I am able to reproduce this with NeRF-SLAM. This is a problem in that repo so please raise an issue there.

closed 2022-12-29 by varunagrawal

VladDok, 2022-12-29:
@varunagrawal, thank for answer.