borglab/gtsam #1365: Add weight factor for GenericProjectionFactorCal3_S2

mattbrown11, 2023-01-03:
First, thanks for developing and supporting such a great library and releasing it as BSD. Copyleft is a deal breaker for my use case, and all other SLAM tools seemed to be copyleft.

## Feature

Add an optional weighting factor to `GenericProjectionFactorCal3_S2` to allow balancing of its influence relative to other factors.

## Motivation

The maximum-likelihood framework employed assumes that the measurements in `GenericProjectionFactorCal3_S2` are  independent and uncorrelated, but this might not be the case, especially for large numbers of image features. If the pose associated with an image is constrained by O(1) inertial factors but then also 100's or 1000's of projection factors, the projection factor can tend to unduly dominate.

For each image, the practitioner you might want use weighting T/N where N is the total number of feature points in the image (number of `GenericProjectionFactorCal3_S2` added) and T is the total visual feature tracking weight applied to the problem for that image. Heuristically, something like T=sqrt(N) might make sense. But, these investigations require the ability to apply a weight.

## Pitch

Add a weighting that scales the log-likelihood contribution for a GenericProjectionFactorCal3_S2

## Alternatives

What I have been doing to actuate this balancing is to add my inertial factors multiple times, e.g., re-add the inertial factor once per image feature factor. But, this seems computationally inefficient.

I also experimented with using the robust loss with really low k. By choice of sigma and k, you can effectively impose L1 error with whatever slope you desire, but L1 loss isn't always appropriate.

I'm not sure what the probabilistic-most-appropriate formulation is for capturing reprojection error correlation within an image. A weight applied to the log-likelihood becomes an exponent scaling of the probability distribution itself, not sure if that makes sense.

varunagrawal, 2023-01-04:
Hi @mattbrown11. From my understanding of your problem, it seems that the optimizer will weigh the projection factors more than the IMU factor, is that correct? If so, would adjusting the stddev values of your noise model be something to consider? You could provide tighter covariances for the `ImuFactor` and looser ones for the `GenericProjectionFactorCal3_S2`.

mattbrown11, 2023-01-04:
Hi varunagrawal, yes, the concern is regarding how the optimizer weighs projection factors versus BetweenFactors. Actually, I think that could work and provides an equivalent result to what I am requesting.

The negative log likelihood (NLL) sum contribution from each `GenericProjectionFactorCal3_S2` is proportional to d^2/sigma^2 where d is a reprojection error (in pixels) and sigma is a measurement certainty (in pixels). So, if you wanted the NLL to be weighted by a scale A, you can just pass sigma/sqrt(A) as the std in `GenericProjectionFactorCal3_S2`. So, if you believe that sigma is the actual image feature measurement standard deviation, in the extreme that you expect all N image-feature errors for an image to be completely correlated, you'd pass in sigma*sqrt(N) into `GenericProjectionFactorCal3_S2`.

closed 2023-01-04 by mattbrown11

varunagrawal, 2023-01-04:
Great! Happy to help.