borglab/gtsam #1378: Add Ubuntu 22.04 release to PPA

ckurtz22, 2023-01-10:
## Feature

Building is incredibly demanding on CPUs. Please add Ubuntu 22.04 release to the PPA to allow users to not have to build manually

## Motivation

Would like to use it in a larger robotics project but no one can set up our workspace because gtsam build consistently causes the setup to crash due to excessive CPU requirements for building.

ProfFan, 2023-01-10:
The build should not crash as long as you don't have too many threads. Have you tried lowering `make -j<N>` to N=2 and 1?

varunagrawal, 2023-01-11:
@berndpfrommer can you please do the needful?

berndpfrommer, 2023-01-11:
@ckurtz22 the nightly packages for the development branch are available for Ubuntu 22.04 (jammy) as well, at least for AMD64 ARM64 and ARMHF:
https://launchpad.net/~borglab/+archive/ubuntu/gtsam-develop/+packages

I just managed to install gtsam from these packages (sorry, not tested yet) on Ubuntu 22.04 x86_64, following these instructions:
https://gtsam.org/get_started/

Does that work for you?

berndpfrommer, 2023-01-19:
Also there is now a ppa for GTSAM v 4.1.1:
https://launchpad.net/~borglab/+archive/ubuntu/gtsam-release-4.1

varunagrawal, 2023-01-21:
@ckurtz22 please let @berndpfrommer know if his suggestions work for you.

ckurtz22, 2023-01-25:
The new 4.1 ppa works, thank you so much!

closed 2023-01-25 by ckurtz22