borglab/gtsam #1382: Incomplete type 'gtsam::HybridGaussianFactorGraph'

varunagrawal, 2023-01-11:
## Description

Compiling a project that uses GTSAM's new hybrid module throws an error

```
../../../include/gtsam/hybrid/HybridNonlinearFactorGraph.h:71:3: error: incomplete type 'gtsam::HybridGaussianFactorGraph' named in nested name specifier
  HybridGaussianFactorGraph::shared_ptr linearize(
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../include/gtsam/hybrid/HybridNonlinearFactorGraph.h:25:7: note: forward declaration of 'gtsam::HybridGaussianFactorGraph'
class HybridGaussianFactorGraph;
```

This error only happens when compiling a project that depends on GTSAM.

The RCA is that the `#include` for `HybridGaussianFactorGraph` was removed and replaced with a forward declaration. The forward declaration works within the GTSAM framework but errors out once we go to an external project.

Tagging @dellaert for his information.

## Steps to reproduce

1. Create a project with a script that uses the `HybridNonlinearFactorGraph`.
2. Compile it with CMake.

## Expected behavior

External project should compile cleanly.

## Environment

## Additional information

varunagrawal, 2023-01-12:
Fix is to include in the .h file (even though the usage is only in the .cpp file).

dellaert, 2023-01-13:
I can add it in next PR

dellaert, 2023-01-13:
> Fix is to include in the .h file (even though the usage is only in the .cpp file).

@varunagrawal maybe I misunderstood: you mean the .h file in your project, and hence it is not a GTSAM issue? In that case I'll close again.

varunagrawal, 2023-01-13:
@dellaert I added the comment in #1383

closed 2023-01-13 by varunagrawal

varunagrawal, 2023-01-15:
@dellaert I am rethinking this issue and I find it odd that a user needs to include `HybridGaussianFactorGraph.h` even when they don't use it. Moreover, this is easy to break since in routine code cleanup, someone may delete that header and then get this weird issue. Thoughts?

dellaert, 2023-01-16:
What happens if you don’t use ::shared_ptr, but use boost::shared_ptr instead?

varunagrawal, 2023-01-16:
That works! Adding it to #1387