borglab/gtsam #1393: What if an outlier is put in NonlinearFactorGraph or ISAM2?

qiaozhijian, 2023-01-19:
Hi, thanks for your work.

Typically, when I use ceres or g2o, I will use robust kernel function first to obtain a not bad state estimation. Then I use this estimation to compute every factor's residual and remove those factors whose residuals are big. Finally, I will optimize the new graph again.

I'm wondering how to do the above operation in NonlinearFactorGraph or ISAM2. Or is there any better solution to removing a wrong factor in graph and optimize again? 

Hope your reply. Thanks very much!

dellaert, 2023-01-21:
Yes, in noiseModel.h and LossFunctions.h there are robust kernels.
Check testNoiseModel.cpp for examples.

closed 2023-01-21 by dellaert

qiaozhijian, 2023-01-29:
> Yes, in noiseModel.h and LossFunctions.h there are robust kernels. Check testNoiseModel.cpp for examples.

Thanks very much! I have found those robust kernels. 

But how do I remove these wrong factors in ISAM? In NonlinearFactorGraph, there is erase function and remove function. I tried the following code, but isam didn't update the state variables.
```
current_estimate = isam.calculateEstimate()
for id in range(factor_graph.nrFactors()):
    if factor_graph.at(id).error(current_estimate) > 10:
        factor_graph.remove(id)
isam.update(factor_graph, initial_estimate)
```
where 10 is a thresold for rejecting outlier.

dellaert, 2023-01-29:
The idea is that with the robust error model you don't actually have to remove them. However, looking through the iSAM header, I remembered that there is an argument in update

```
* @param removeFactorIndices Indices of factors to remove from system
```

and a test 

```c++
TEST(ISAM2, removeVariables)
{
  // These variables will be reused and accumulate factors and values
  Values fullinit;
  NonlinearFactorGraph fullgraph;
  ISAM2 isam = createSlamlikeISAM2(fullinit, fullgraph, ISAM2Params(ISAM2GaussNewtonParams(0.001), 0.0, 0, false));

  // Remove the measurement on landmark 0 (Key 100)
  FactorIndices toRemove;
  toRemove.push_back(7);
  toRemove.push_back(14);
  isam.update(NonlinearFactorGraph(), Values(), toRemove);

  // Remove the factors and variable from the full system
  fullgraph.remove(7);
  fullgraph.remove(14);
  fullinit.erase(100);

  // Compare solutions
  CHECK(isam_check(fullgraph, fullinit, isam, *this, result_));
}
```

in `testGaussianISAM2.cpp`.

Does that work for you?

qiaozhijian, 2023-01-30:
Thank you for your reply. I find that graph optimization will become slower if I use robust error model. And I expect to remove these outliers to obtain a better estimation. Keeping these outliers in the graph makes me feel uneasy...

I have tried codes in testGaussianISAM2.cpp, but I transfered them to python.
```
parameters = gtsam.ISAM2Params()
parameters.setRelinearizeThreshold(0.0) # default 0.1
parameters.relinearizeSkip = 0 # default 10
parameters.enableRelinearization = False # default true
isam = gtsam.ISAM2(parameters)
//(Construct factor_graph and isam.update(*,*) are omitted)
current_estimate = isam.calculateEstimate()
toRemove = []
for id in range(factor_graph.nrFactors()):
    if factor_graph.at(id).error(current_estimate) > 10:
        toRemove.append(id)
isam.update(gtsam.NonlinearFactorGraph(), gtsam.Values(), toRemove)
```
However, the problem is that isam didn't update the variables.

dellaert, 2023-01-30:
I think it's a good idea to add a python version. Can you attach the entire (failing) test in a file test_GaussianISAM2.py? I'm intrigued because the wrapper just calls the underlying c++, so if the c++ passes the python test should pass. But maybe the c++ is also failing but not adequately tested, so I'm also interested in having your test for the non-removed variables.

qiaozhijian, 2023-01-30:
> I think it's a good idea to add a python version. Can you attach the entire (failing) test in a file test_GaussianISAM2.py? I'm intrigued because the wrapper just calls the underlying c++, so if the [...]

Thanks! Previously, I used GTSAM to fit a spline. To make this demo simpler, I just estimate a 3d point. The code is following,
```
#!/usr/bin/env python
import numpy as np
from gtsam.symbol_shorthand import P
import gtsam

if __name__ == '__main__':

    np.random.seed(1234)
    point3d = np.array([-2, 1, 0])

    # We now can use nonlinear factor graphs
    factor_graph = gtsam.NonlinearFactorGraph()
    parameters = gtsam.ISAM2Params()
    parameters.setRelinearizeThreshold(0.0) # default 0.1
    parameters.relinearizeSkip = 0 # default 10
    parameters.enableRelinearization = False # default true

    isam = gtsam.ISAM2(parameters)
    # New Values container
    initial_estimate = gtsam.Values()
    # Add initial estimates to the Values container
    initial_estimate.insert(P(0), gtsam.Point3(point3d))

    # Add factors for catmull measurements
    pt_noise_sigma = 0.1
    pt_noise_model = gtsam.noiseModel.Isotropic.Sigma(3, pt_noise_sigma)
    # pt_noise_model = gtsam.noiseModel.Robust.Create(gtsam.noiseModel.mEstimator.Huber(0.5), pt_noise_model)
    outlier_idx = []
    for k in range(50):
        if (k + 1) % 10 == 0:
            outlier = point3d + np.random.normal(0, 10, size=point3d.shape)
            gf = gtsam.PriorFactorPoint3(P(0), gtsam.Point3(outlier), pt_noise_model)
            outlier_idx.append(factor_graph.size())
        else:
            meas_pt = point3d + np.random.normal(0, pt_noise_sigma, size=point3d.shape)
            gf = gtsam.PriorFactorPoint3(P(0), gtsam.Point3(meas_pt), pt_noise_model)
        factor_graph.add(gf)
        isam.update(factor_graph, initial_estimate)
        if k == 0:
            initial_estimate.clear()
    current_estimate = isam.calculateEstimate()
    print('true estimate: ', point3d)
    print('current estimate: ', current_estimate.atPoint3(P(0)))

    toRemove = []
    for id in range(factor_graph.nrFactors()):
        # print('error {}: '.format(id), factor_graph.at(id).error(current_estimate))
        if factor_graph.at(id).error(current_estimate) > 1000:
            toRemove.append(id)
    print('outlier_idx: ', outlier_idx)
    print('toRemove: ', toRemove)
    toRemove = gtsam.KeyVector(toRemove)
    isam.update(gtsam.NonlinearFactorGraph(), initial_estimate, toRemove)
    updated_estimate = isam.calculateEstimate()
    print('updated estimate: ', updated_estimate.atPoint3(P(0)))
```
My output:
```
true estimate:  [-2  1  0]
current estimate:  [-2.56725811  1.101162    1.01802243]
outlier_idx:  [9, 19, 29, 39, 49]
toRemove:  [9, 19, 29, 39, 49]
updated estimate:  [-2.56926374  1.10175421  1.02185693]
```

v-diepttn147, 2024-12-13:
Hello @qiaozhijian, I also encountered this problem, which I want to remove outliers out of the graph using `graph.remove(index)` but the system seemed to still optimize with those outliers, have you solved this issues? Or do you have work around solution? Thank you