borglab/gtsam #1400: Build failing: ‘class gtsam::Values’ has no member named ‘filter’

husseinalijaafar, 2023-01-21:
## Description

Trying to build GTSAM with the depreciated flag below, but the build is failing. Both myself and a colleague are having similar issues.  
```
[ 84%] Building CXX object examples/CMakeFiles/StereoVOExample_large.dir/StereoVOExample_large.cpp.o
/home/hussein/gbp/gtsam/examples/StereoVOExample_large.cpp: In function ‘int main(int, char**)’:
/home/hussein/gbp/gtsam/examples/StereoVOExample_large.cpp:116:31: error: ‘class gtsam::Values’ has no member named ‘filter’
  116 |   Values pose_values = result.filter<Pose3>();
      |                               ^~~~~~
/home/hussein/gbp/gtsam/examples/StereoVOExample_large.cpp:116:43: error: expected primary-expression before ‘>’ token
  116 |   Values pose_values = result.filter<Pose3>();
      |                                           ^
/home/hussein/gbp/gtsam/examples/StereoVOExample_large.cpp:116:45: error: expected primary-expression before ‘)’ token
  116 |   Values pose_values = result.filter<Pose3>();
      |                                             ^
```
## Steps to reproduce

```
git clone https://github.com/borglab/gtsam.git
cd gtsam
mkdir build && cd build
cmake -DGTSAM_ALLOW_DEPRECATED_SINCE_V42:=OFF .. 
sudo make install
```

## Expected behavior

Build successfuly

## Environment

Xubuntu 20.04, cmake version 3.24.0-rc3

## Additional information

I had previously successfully built gtsam a few days ago. This may be due recent changes.

dellaert, 2023-01-21:
Thanks! Fixed in #1399 in develop which should be merged once reviewed by @jlblancoc or @varunagrawal , but maybe you are trying to compile release/4.2 ? I should make the changes there, too, PR coming!

closed 2023-01-21 by varunagrawal

husseinalijaafar, 2023-01-21:
Thank you for the prompt responses Dr!