borglab/gtsam #1412: ubuntu nightly build fails with error: ‘library_version_type’ is not a member of ‘boost::serialization’

berndpfrommer, 2023-01-24:
```
/<<PKGBUILDDIR>>/gtsam/3rdparty/CCOLAMD/Include -isystem /usr/include/eigen3 -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -g -DNDEBUG -fPIC -Wall -fPIC -Wreturn-local-addr -Werror=return-local-addr -Wreturn-type -Werror=return-type -Wformat -Werror=format-security -Wsuggest-override -g -O3 -Wno-unused-local-typedefs -MD -MT gtsam/CMakeFiles/gtsam.dir/inference/BayesTree.cpp.o -MF CMakeFiles/gtsam.dir/inference/BayesTree.cpp.o.d -o CMakeFiles/gtsam.dir/inference/BayesTree.cpp.o -c /<<PKGBUILDDIR>>/gtsam/inference/BayesTree.cpp
In file included from /<<PKGBUILDDIR>>/gtsam/base/FastList.h:27,
                 from /<<PKGBUILDDIR>>/gtsam/inference/Key.h:20,
                 from /<<PKGBUILDDIR>>/gtsam/inference/BayesTree.h:24,
                 from /<<PKGBUILDDIR>>/gtsam/inference/BayesTree.cpp:21:
/usr/include/boost/serialization/list.hpp: In function ‘void boost::serialization::load(Archive&, std::__cxx11::list<_ValT, _Allocator>&, unsigned int)’:
/usr/include/boost/serialization/list.hpp:53:33: error: ‘library_version_type’ in namespace ‘boost::serialization’ does not name a type; did you mean ‘item_version_type’?
   53 |     const boost::serialization::library_version_type library_version(
      |                                 ^~~~~~~~~~~~~~~~~~~~
      |                                 item_version_type
/usr/include/boost/serialization/list.hpp:60:30: error: ‘library_version_type’ is not a member of ‘boost::serialization’; did you mean ‘item_version_type’?
   60 |     if(boost::serialization::library_version_type(3) < library_version){
      |                              ^~~~~~~~~~~~~~~~~~~~
      |                              item_version_type
/usr/include/boost/serialization/list.hpp:60:56: error: ‘library_version’ was not declared in this scope
   60 |     if(boost::serialization::library_version_type(3) < library_version){
      |                                                        ^~~~~~~~~~~~~~~
make[3]: *** [gtsam/CMakeFiles/gtsam.dir/build.make:695: gtsam/CMakeFiles/gtsam.dir/inference/BayesTree.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
```

dellaert, 2023-01-24:
You might also need #include <boost/version.hpp> to do so.

berndpfrommer, 2023-01-25:
Apologies for this terrible bug report. I didn't even say what OS the problem was on: Ubuntu 22.04. The 
 other Ubuntu versions actually work.

For the last few days the Ubuntu packaging has failed. The way Ubuntu packaging works is that you have to upload the original sources (GTSAM) and the packaging instructions ("debian" folder) to Canonical's servers, which then compile the source for you. This compilation fails for Ubuntu 22.04.
It turns out that GTSAM's continuous integration (CI) testing does not yet cover Ubuntu 22.04.

Despite apparently not being covered by automated tests, until a few days ago GTSAM was compiling on Ubuntu 22.04 as well, but some of the last few commits (looks like a laudable effort is made to reduce dependencies on boost) broke things on Ubuntu 22.04

Here is a link to the full build log that includes the error message above:
https://launchpadlibrarian.net/647725382/buildlog_ubuntu-jammy-amd64.gtsam-no-tbb_4.1.0-1ubuntu2~1589.gbp173322_BUILDING.txt.gz

I will dig in and see if I can reproduce the problems on my own machine (running Ubuntu 22.04) and try your suggestions above.

berndpfrommer, 2023-01-25:
Could be related to this: https://github.com/boostorg/serialization/issues/219

berndpfrommer, 2023-01-25:
I submitted a PR that fixes this issue. If you are already submitting a fix, please just ignore/reject the PR.

closed 2023-01-25 by berndpfrommer