borglab/gtsam #1414: Status of removing boost from gtsam

kartikarcot, 2023-01-24:
# Describing: `kartikarcot:verdant/boost-misc`

As of today, this branch is the one with the least of amount of boost in it. The branch however needs a bit of work to fix some things:

- [ ] Ordering.h has this instance that needs to be fixed:
```
  /// Add new variables to the ordering as ordering += key1, key2, ...  Equivalent to calling
  /// push_back.
  boost::assign::list_inserter<boost::assign_detail::call_push_back<This> > operator+=(
      Key key) {
    return boost::assign::make_list_inserter(
        boost::assign_detail::call_push_back<This>(*this))(key);
  }
```
- [ ] GnC Optimizer was deactivated because there was no easy replacement for `Chi2Inv`. It was deactivated in both
python bindings as well as in Cpp code. It will have to be brought back.

- [ ] Python bindings DONT COMPILE because `EliminateAbleFactorGraph` moved away fram `boost::variant` to `std::variant`. This in turn led to moving to using `std::reference_wrapper`. Which causes issues with python bindings. TLDR version is:
```
// A call that used to be like this:
marginalMultiFrontalBayesNet(ordering,....)
// Now has to be called like this:
marginalMultiFrontalBayesNet(std::cref(ordering),....)
// And this does not play well with the wrap library
```
- [ ] CppUnitLite library uses `boost::lexical_cast` that needs to be replaced with `std::to_string` maybe
- [ ] All timing functionality provided by `timing.h` were conditionally compiled away since `boost::timers` did not have an easy stand-in replacement
- [ ] `Scheduler` and it's examples and it's tests were disabled because it used boost::tokenizer and boost::escaped_list_seperator
- [ ] `QPSolver` and it's examples, and it's tests were disabled
- [ ] `FindSeperator-inl.h` has usage of `boost::shared_array` that needs to be replaced or deactivated

## Other things

Apart from this there is a bunch of stuff in Values.h regarding iterators and graphs which use boost's graphing features that need to be replaced as well.

ProfFan, 2023-01-24:
@dellaert I think the custom operators should be removed, as `Ordering` moves to being a std container

dellaert, 2023-06-04:
This is done :-)

closed 2023-06-04 by dellaert