borglab/gtsam #1436: Double free or corruption (out) error

enahon, 2023-02-03:
## Description

I am linking GTSAM to an executable which also uses PCL and Eigen. When I launch the executable, the program immediately crashes with the following error: "Double free or corruption (out)".
Tha call stack indicates the following:

libc.so.6!__pthread_kill_implementation(int no_tid, int signo, pthread_t threadid) (pthread_kill.c:44)
libc.so.6!__pthread_kill_internal(int signo, pthread_t threadid) (pthread_kill.c:78)
libc.so.6!__GI___pthread_kill(pthread_t threadid, int signo) (pthread_kill.c:89)
libc.so.6!__GI_raise(int sig) (raise.c:26)
libc.so.6!__GI_abort() (abort.c:79)
libc.so.6!__libc_message(enum __libc_message_action action, const char * fmt) (libc_fatal.c:155)
libc.so.6!malloc_printerr(const char * str) (malloc.c:5664)
libc.so.6!_int_free(mstate av, mchunkptr p, int have_lock) (malloc.c:4588)
libc.so.6!__GI___libc_free(void * mem) (malloc.c:3391)
libgtsam.so.4!_GLOBAL__sub_I_lago.cpp (Unknown Source:0)
ld-linux-x86-64.so.2!call_init(struct link_map * l, int argc, char ** argv, char ** env) (dl-init.c:70)
ld-linux-x86-64.so.2!call_init(char ** env, char ** argv, int argc, struct link_map * l) (dl-init.c:33)
ld-linux-x86-64.so.2!_dl_init(struct link_map * main_map, int argc, char ** argv, char ** env) (dl-init.c:117)
ld-linux-x86-64.so.2!_dl_start_user (Unknown Source:0)
[Unknown/Just-In-Time compiled code] (Unknown Source:0)

## Environment

I am running this on Ubuntu 22.04 and using GCC 11.3.0. I have built GTSAM v4.1.1 and PCL 1.13.0 from source. I am also using Eigen 3.4.0.

I have built GTSAM using the following options:
        -DGTSAM_USE_SYSTEM_EIGEN=ON
        -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF
I tried enabling DGTSAM_BUILD_WITH_MARCH_NATIVE but the same error occurs. Upgrading GTSAM to the latest 4.2a9 pre-release didn't help either.
I have also tried adding the following option when building GTSAM: 
        -DCMAKE_CXX_FLAGS:STRING=-std=c++17

I am building PCL with default options.

Any ideas will be much appreciated!
Thanks.

## Additional information

dellaert, 2023-02-04:
Dang, that's a hard one. Can you try with version 4.2a9?

enahon, 2023-02-04:
Unfortunately I have already tried and it didn't really help. The stack trace above is actually from 4.2a9

dellaert, 2023-02-04:
Unfortunately I'm no great help here, maybe @ProfFan can help.
One other thing to try is GTSAM_USE_SYSTEM_EIGEN=OFF

enahon, 2023-02-04:
Thanks but also tried that as I mentioned above and didn't help either...

thorstink, 2023-02-05:
I also experienced this if the march-flag was set differently for different libraries. In our production setup we compile all the libraries that use eigen from source with the same compilation settings: cpp17 to avoid eigen alignment issues and the same march-flag.If you compile your project with address sanitizer, and it points to some eigen function where you typically experience this double free, I think the above suggestion might help you. Thomas

ProfFan, 2023-02-06:
I think we should convert this to discussion. @thorstink 's opinion is what I think is most likely to be the case. Remember that you have to make *ABSOLUTELY SURE* that your libraries are using the same settings for Eigen.