borglab/gtsam #1446: Why the optimizer didn't work and the deltaTij_ is zero?

JiangWeiHn, 2023-02-06:
I change the ImuFactorsExample.cpp to  fit my dataset.It works well  at the IMU preintegration part. Bug when use LevenbergMarquardtOptimizer. I find that the optimizer didn't work,the result exactly equal to the prop_state.After debug the programme, I find that when calculate the ImuFactor error ,the  deltaTij is zero.And the error is inf.But the preintegration part the deltaTij is not zero.It confuse me . Please help me ,I donot know how to fix it.Here is my code:
      GPSFactor gps_factor(X(correction_count),gps, // D,
                           correction_noise);
      graph->add(gps_factor);

      // Now optimize and compare results.
      prop_state = preintegrated->predict(prev_state, prev_bias);
        
      initial_values.insert(X(correction_count), prop_state.pose());
      initial_values.insert(V(correction_count), prop_state.v());
      initial_values.insert(B(correction_count), prev_bias);
      Values result;
      if (use_isam) {
        isam2->update(*graph, initial_values);
        result = isam2->calculateEstimate();

        // reset the graph
        graph->resize(0);
        initial_values.clear();

      } else {
        LevenbergMarquardtOptimizer optimizer(*graph, initial_values);
        result = optimizer.optimize();
      }
       
      prev_state = NavState(result.at<Pose3>(X(correction_count)),
                            result.at<Vector3>(V(correction_count)));
      prev_bias = result.at<imuBias::ConstantBias>(B(correction_count));

dellaert, 2023-02-06:
Where is the IMUFactor?

JiangWeiHn, 2023-02-07:
sorry ,i didn't paste all the code.This is it.

#include <boost/program_options.hpp>

// GTSAM related includes.
#include <gtsam/inference/Symbol.h>
#include <gtsam/navigation/CombinedImuFactor.h>
#include <gtsam/navigation/GPSFactor.h>
#include <gtsam/navigation/ImuFactor.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/slam/dataset.h>
#include <gtsam/navigation/OdometerFactor.h>
#include <gtsam/navigation/ImuFactor.h>
#include <cstring>
#include <fstream>
#include <iostream>
#include <cmath>

#define PI 3.141592653589793
#define R2D 57.295779513082320876798154814105f
#define D2R 0.017453292519943295769236907684886f

using namespace gtsam;
using namespace std;

using symbol_shorthand::B;  // Bias  (ax,ay,az,gx,gy,gz)
using symbol_shorthand::V;  // Vel   (xdot,ydot,zdot)
using symbol_shorthand::X;  // Pose3 (x,y,z,r,p,y)
using symbol_shorthand::A;  // DeltaAngle Vector3(r,p,y)
using symbol_shorthand::M;  //DeltaArm Vector3(x,y,z)

// namespace po = boost::program_options;
double starttime = 440190;
double endtime = 442050;
struct ImuRaw {
    double sow;       /* s */
    double dtheta[3]; /* gyro, rad */
    double dvel[3];   /* acce, m/s */
};

boost::shared_ptr<PreintegratedCombinedMeasurements::Params> imuParams(double imuGravity) {
  // We use the sensor specs to build the noise model for the IMU factor.
  double accel_noise_sigma = 0.05;
  double gyro_noise_sigma = 0.001;
  double accel_bias_rw_sigma = 0.005;
  double gyro_bias_rw_sigma = 0.0008;
  Matrix33 measured_acc_cov = I_3x3 * pow(accel_noise_sigma, 2);
  Matrix33 measured_omega_cov = I_3x3 * pow(gyro_noise_sigma, 2);
  Matrix33 integration_error_cov =
      I_3x3 * 1e-4;  // error committed in integrating position from velocities
  Matrix33 bias_acc_cov = I_3x3 * pow(accel_bias_rw_sigma, 2);
  Matrix33 bias_omega_cov = I_3x3 * pow(gyro_bias_rw_sigma, 2);
  Matrix66 bias_acc_omega_init =
      I_6x6 * 1e-4;  // error in the bias used for preintegration

  auto p = PreintegratedCombinedMeasurements::Params::MakeSharedD(imuGravity);
  // PreintegrationBase params:
  p->accelerometerCovariance =
      measured_acc_cov;  // acc white noise in continuous
  p->integrationCovariance =
      integration_error_cov;  // integration uncertainty continuous
  // should be using 2nd order integration
  // PreintegratedRotation params:
  p->gyroscopeCovariance =
      measured_omega_cov;  // gyro white noise in continuous
  // PreintegrationCombinedMeasurements params:
  p->biasAccCovariance = bias_acc_cov;      // acc bias in continuous
[... 271 lines cut ...]
      // allOd = 0;
      // prior_attitude_angle = AttitudeAngle;
      // // Reset the preintegration object.
      // prior_position = result.at<Pose3>(X(correction_count)).translation();
      ifgnss>>gnsst>>now(0)>>now(1)>>now(2)>>Rk(0)>>Rk(1)>>Rk(2);
    }
    if (tk>odot){
      // thetaCurrent  =  preint_imu.preintegrated().head(3)-prev_state.pose().rotation().rpy();
      // odpreintegrated->OdIntegrateMeasurement(measuredOd,InstallAngle,rk,S0,
      //                                       thetaCurrent,thetaPrior,arm);
      // thetaPrior = thetaCurrent;
      // ifodo>>odot>>odovl>>odovr;
      // measuredOd = (odovl+odovr)/2.0*0.1;
      // allOd +=measuredOd;

     
    }

  }
  fclose(fp_out);
  fclose(fd_out);
  cout << "Complete, results written to " << output_filename << "\n\n";

  return 0;
}

JiangWeiHn, 2023-02-07:
when i try the isam2,there is a error:
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.

JiangWeiHn, 2023-02-07:
I print the initial_values,it's this:
Values with 6 values:
Value b0: (gtsam::imuBias::ConstantBias)
acc = 0 0 0 gyro = 0 0 0

Value b1: (gtsam::imuBias::ConstantBias)
acc = 0 0 0 gyro = 0 0 0

Value v0: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        0;
        0;
        0
]

Value v1: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        -0.773251;
        0.0277616;
        0.039793
]

Value x0: (gtsam::Pose3)
R: [
        -0.995791, -0.0914287, -0.00633581;
        0.0914909, -0.995752, -0.0103532;
        -0.00536232, -0.0108893, 0.999926
]
t: 0 0 0

Value x1: (gtsam::Pose3)
R: [
        -0.995054, -0.098555, -0.0124033;
        0.0986521, -0.995094, -0.00747505;
        -0.0116057, -0.00866169, 0.999895
]
t:   -0.155385 0.000114898   0.0186261

JiangWeiHn, 2023-02-07:
I find my fault, i put the preint_imu in the begin，and it point to default PreintegratedImuMeasurements,the preintMeasCov is zero.thanks a lot for help!

closed 2023-02-07 by JiangWeiHn